nefarius/ScpToolkit #3: Screensaver will not start (SCPServer, Windows 8.1)

CrazySteve88, 2015-08-16:
For some reason the SCPServer service will prevent screen savers from starting, at least with Windows 8.1.

The same issue occurs with the original SCP drivers. After enabling the SCPService, screen savers stop working. As soon as I disable the service everything works fine. Again this is an issue that's persisted since the original drivers.

I installed the service, ps3 drivers, I don't use Bluetooth. I looked at the logs but I didn't notice anything strange.

nefarius, 2015-08-28:
I can confirm the behavior, I'll look into that!

brampower, 2015-09-09:
Just wanted to pitch in by saying I have the same issue. If there are any logs required (or anything else), just let me know!

StephenHey, 2015-10-30:
I am also having this problem in Windows 10.  Sleep mode/screen saver/turning off monitors all do not work unless you stop the service.  Stopping the service with a ps3 controller plugged in starts the ps3, which is annoying.  I'd like to not have to stop the service :)

nefarius, 2015-11-29:
I tried swapping out the deprecated timer code but it didn't make a difference, will continue to investigate =/

CrazySteve88, 2015-11-29:
I noticed DS4 Windows doesn't do this and it uses the same setup as the SCP
Drivers. You can find the source here:
https://github.com/Jays2Kings/DS4Windows/releases/tag/v1.4.3

Not sure if it'll help much though, but hopefully this can be fixed.

nefarius, 2015-11-30:
DS4Windows has a completely different working design and only uses a small portion of the Scarlet code, I've already looked into it.

nefarius, 2015-12-12:
Just FYI it happens on Windows 10 too sadly =/

josefrvaldes, 2016-01-04:
Yes, it happens to me too in W10.

rkubes, 2016-02-12:
I have Windows 10 and noticed this issue. The difference for me is that it only affects the Screen Saver - it does not affect the Video Off function.

I have the following settings:
ScreenSaver - 1 minute
Video Off - 5 minutes
Sleep - Never (have a RAID controller that doesn't support it)

Have not tested with a DS4; however, with a DS3 controller if it is plugged in and on, or if it is on via Bluetooth, then the Screensaver will not activate. That being said, the Video Off will still work at 5 minutes even though the Screensaver was blocked.

If I turn off the DS3 controller - without having to touch the SCP service - the Screen Saver will work again.

The frustration for me is that I have Windows require a password on wakeup; however, since the Video is going off without the ScreenSaver running, it will not ask for a password when woken up.

spacetraveller, 2017-06-11:
I'm having this problem, even with the latest releases.

nefarius, 2017-06-11:
@spacetraveller that's because it hasn't been fixed yet.

rgaber, 2017-09-06:
I see you id'd the problem, so when do you plan on releasing this?

closed 2018-01-26 by nefarius