nefarius/ScpToolkit #5: Controller Compatibility

vurso, 2015-08-19:
Hi,

I have an after market controller with the following id:  

USB\VID_045E&PID_028E&REV_0114
USB\VID_045E&PID_028E

I am assuming its not currently compatible?

Thanks,

V

nefarius, 2015-08-22:
You may have luck, just add the HIDs to `ScpControl.ini ` and run the installer again.

xmakina, 2015-09-07:
I didn't have any luck with this, but the controller used to work in a previous version of ScpControl. Is there any way to add compatibility back in?

nefarius, 2015-09-07:
Do you know at which version/commit is stopped working?

tevenan, 2015-09-25:
I have the same problem.
Controller HID is the same as above (USB\VID_045E&PID_028E&REV_0114 and USB\VID_045E&PID_028E)

BT Dongle HID is:
USB\VID_0A12&PID_0001&REV_8891
USB\VID_0A12&PID_0001

Both were already listed in the ScpControl.ini, I tried adding the "REV_0114 and REV_8891" with no luck.

This combination works on my laptop using ScpService.exe 1.2.2.175 and the dll posted here: 
http://www.filedropper.com/scpcontrol

Unfortunately I haven't been able to replicate it on any other computer. The SCP DS3 Service fails if I try to run it with the modified dll on all versions of ScpService.exe.

Additional information:
- Without the modified dll, the DS3 controller flashes as though it's connecting to bluetooth, then the lights turn off, the controller stops responding, and requires a hard reset.

greengros, 2017-01-24:
So, I had the same issue,  and i tried to add my controller HID to the list, but i could not replace the old one because i get a message stating "Access is Denied".

closed 2018-01-26 by nefarius