nefarius/ScpToolkit #6: Sixaxis API extension?

CookiePLMonster, 2015-08-27:
Since SCP's xinput DLL already expands the API with XInputGetExtended to poll pressure sensitive buttons, any chance to see another API extension allowing to poll Sixaxis data?

I have already done that with older SCP by making my application inject custom code into SCP xinput DLL on runtime... so surely a legit API extension would be a more sensible idea. Could easily provide the coding for that and issue a merge request, would just need to know it wouldn't be called 'useless'.

Cheers!

Blackbird88, 2015-08-27:
Yes please. SixAxis is great future to have. PS3 emulators might also benefit from it :)

YogSottot, 2015-08-28:
It also would be useful in Dolphin for emulation wii-remote.

nefarius, 2015-09-07:
Code would be very much appreciated. I guess by sixaxis you don't refer to the predecessor of the DS3 but the gyroscope and accelerometer features?

CookiePLMonster, 2015-09-07:
Yep, I mean accelerometers and a gyroscope, of course.

Also do note I only can test DS3 and for some reason I only can test wired. Even though BTH driver installs properly, I could never pair my pad with it - neither with old SCP or with the one maintained by you. All I could find on this issue was "sorry, your controller is not original" - though seeing how mine came with a PS3...

nefarius, 2015-10-06:
Soooo... no code to contribute?

CookiePLMonster, 2015-10-06:
I'd probably need something to test it with, is ScpUser xinput1_3.dll working again or not yet?

Also, I'm actually still on Scarlett's driver, I had plenty of weird issues with your one... mostly the pad working only with XInput and not through DirectInput etc.

mirhl, 2015-10-07:
Just for the records, [here](http://gtaforums.com/topic/562765-iiivcsaasi-ginput/?p=1066552403)'s the various info.

nefarius, 2015-10-07:
Test? I though't there's working code?

CookiePLMonster, 2015-10-07:
Kind of. It's a hack, injecting code straight into latest Scarlett's ScpUser xinput1_3.dll on runtime.

CookiePLMonster, 2015-10-31:
I see pressure sensitivity, does it mean ScpUser is working again? :D

Also got a BT dongle so it's a great excuse to finally move from Scarlett's version on your version. Can't test the dongle now but as soon as I get it to work (or not) I'll list it in compat dongles list, provided it's not there yet.

nefarius, 2015-10-31:
Yes it's working again, although it's not really the ScpUser project fixed, it's a re-write from scratch. Please see my Wiki for more details.

CookiePLMonster, 2015-10-31:
I noticed but ughh, these issues will never stop...with this new BTH dongle (in compatible devices list again) only analog sticks and face buttons work, the rest is broken. ffs.

nefarius, 2015-11-29:
Since this is in the works and the issue was kinda hijacked anyways I'm closing this one :)

closed 2015-11-29 by nefarius