nefarius/ScpToolkit #7: Compatible bluetooth dongles?

LinkTree3, 2015-09-07:
Hi! firstly I'd like to thank you for continuing scarlet's work. His latest release didn't run on my windows 7 for some reason so I am glad I managed to find your post and work.

I came to ask you about the compatibility of USB bluetooth dongles, I've seen your device ID list but that doesn't really help me if I am looking to buy a compatible dongle since sellers don't list that data and most of them wouldn't even know about USB devices IDs if asked about it.

Would this dongle work with the program?
http://www.ebay.com/itm/Mini-USB-Bluetooth-V4-0-Dongle-Wireless-Adapter-Device-New/181651319031?_trksid=p2045573.c100033.m2042&_trkparms=aid%3D111001%26algo%3DREC.SEED%26ao%3D1%26asc%3D33277%26meid%3D56898a0ab6eb453790075385fc5a534e%26pid%3D100033%26rk%3D3%26rkt%3D4%26sd%3D221534853709

Thanks!

nefarius, 2015-09-09:
If it's genuine and has v4 specification it should work fine!

LinkTree3, 2015-10-04:
I received the dongle I ordered and it's VID/PID is this:
USB\VID_0A12&PID_0001

It appears on the list as compatible and in my device manager it appears under libusbk along with my Sony DS3 but for some reason they wont pair with each other... Maybe I am doing something wrong?

LinkTree3, 2015-10-05:
I took a few images that might help
http://imgur.com/a/vJSeD

LinkTree3, 2015-10-09:
I still have the same issue. By the way, the controller receives rumble requests from ScpServer.exe over bluetooth and usb.

I tried uninstalling and reinstalling and reseting my controller multiple times. I have no idea what has changed that is causing it not to work now not with USB connection and not with bluetooth.

Please suggest me what to do. At this point I don't even care about bluetooth I just want to get it back to work over USB.

LinkTree3, 2015-10-12:
This was solved by uninstalling and reinstalling the microsoft xbox controller driver.

closed 2015-10-12 by LinkTree3