nefarius/ScpToolkit #10: Driver Overwriting Integrated Bluetooth Controller

Chosen-One, 2015-09-10:
So I was looking for a PS3 controller driver that worked in Windows 10, and ran into this.  I used to run the original ScpServer so I'm mostly familiar with how this thing works.  Problem is, when installing, it overwrites not only the Bluetooth Dongle, but the Bluetooth radio built into my laptop (even though I disabled it in the device manager to avoid this issue). Oddly enough, this issue did not arise when using the original version of ScpServer. Luckily, uninstalling the original Bluetooth radio and checking the uninstall driver software option, then allowing Windows to reinstall the device, seems to have reverted it back to its original state.

Here are the IDs of both Bluetooth radios:

Integrated Radio:
Device Description: Intel(R) Wireless Bluetooth(R)
Hardware IDs:
USB\VID_8087&PID_07DC&REV_0001
USB\VID_8087&PID_07DC

USB Bluetooth Dongle:
Device Description: CSR8510 A10
Hardware IDs:
USB\VID_0A12&PID_0001&REV_8891
USB\VID_0A12&PID_0001

Chosen-One, 2015-09-11:
Did a reboot and the Integrated Radio went back to using the Scp Driver. Installing official driver for the Integrated Radio to see if that solves the problem.

nefarius, 2015-11-29:
This is solved in the latest version.

closed 2015-11-29 by nefarius