nefarius/ScpToolkit #11: DS3 vibration aint working over bluetooth

volvah, 2015-09-13:
Sup,
So as title says im using DS3 (not genuine one) via bluetooth dongle and the issue is that rumble is broken when wireless (just rumble, controller working fine). Thats not a big issue but a man can dream, right?

So heres my log from ScpServer dot exe http://i.imgur.com/LtNvMfx.png (ye i know thats silly but i dunno how to export this log in text or something) and also part of log when i plug controller using usb http://i.imgur.com/sjA4DVT.png (rumble working when usb).

Like, maybe theres some workaround to fix this one, thanks.

nefarius, 2015-09-15:
The log is written to files. Please look in the folder you extracted the program in and you should find some `*.log.xml` files, submit those and no screenshots needed :)

volvah, 2015-09-15:
Ok, well, some xml logs:
ScpDsxServer.log.xml - http://pastebin.com/FttfZAYW
ScpDsxService.log.xml - http://pastebin.com/3Jh0gzSd

gjveron, 2015-09-25:
i second this issue. rumble works via cable but not over bluetooth on a fake ds3.

ScpDsxServer.log.xml - http://pastebin.com/aEjjLWK8

volvah, 2015-10-24:
Seems like your latest release fixed this for me, good job mate!

closed 2015-10-24 by volvah