nefarius/ScpToolkit #12: Non original controller ds3 not pairing bluetooth

lfavaro, 2015-09-17:
Hi, i am trying to pairing my controller (not sony) with this ID:
USB\VID_054C&PID_0268&REV_0100
USB\VID_054C&PID_0268
With Usb, it works fine.
My bluetooth adapter is an integrated laptop one : 
USB\VID_04CA&PID_300B&REV_0001
USB\VID_04CA&PID_300B
Using the scpdriver.exe , the program identify the card and install the driver without problem. 
I send you the log, see the latest 10 rows: http://pastebin.com/84PDjEsb
Thank you in advance

lfavaro, 2015-09-19:
Edit: i just add the bt remote name of my controller in the scpcontrol.ini like this:
[HCI]
; DualShock 3 (partial) product name
SupportedName = PLAYSTATION(R)3
; ???
SupportedName = Navigation Controller
; DualShock 4 product name
SupportedName = Wireless Controller
; My contrller
SupportedName = RDA bt

Now the log has changed but still not connecting http://pastebin.com/E9pRVrh2:

PetarT, 2015-11-10:
Any luck with this @nefarius ? :)
This is log I get with same issue as @lfavaro mention.
It tries to connect to my bt but it just fails afterwards.
[image]

closed 2018-01-26 by nefarius