nefarius/ScpToolkit #13: win 10 pro simplified Chinese

sxd1140, 2015-09-21:
2015/9/21 23:43:30 [INFO] - SCP Driver Installer 1.5.5740.18577 [09/21/2015 23:43:30]
2015/9/21 23:43:30 [DEBUG] - Preparing to load DIFxAPI
2015/9/21 23:43:30 [INFO] - Running as 64-Bit process
2015/9/21 23:43:30 [DEBUG] - DIFxAPI path: C:\xx\bin\DIFxApi\amd64\DIFxAPI.dll
2015/9/21 23:43:30 [DEBUG] - Loaded library: C:\xx\bin\DIFxApi\amd64\DIFxAPI.dll
2015/9/21 23:43:30 [INFO] - Microsoft Windows 10 (6.2.9200.0 AMD64) detected
2015/9/21 23:43:30 [INFO] - Selected Win10 configuration
2015/9/21 23:43:34 [INFO] - ENTER:  DriverPackageInstallW
2015/9/21 23:43:34 [INFO] - Looking for Model Section [SCProductions.NTamd64]...
2015/9/21 23:43:34 [INFO] - RETURN: DriverPackageInstallW  (0x103)
2015/9/21 23:43:34 [DEBUG] - Preparing to load libwdi
2015/9/21 23:43:34 [INFO] - Running as 64-Bit process
2015/9/21 23:43:34 [DEBUG] - libwdi path: C:\xx\bin\libwdi\amd64\libwdi.dll
2015/9/21 23:43:34 [DEBUG] - Loaded library: C:\xx\bin\libwdi\amd64\libwdi.dll
2015/9/21 23:43:34 [ERROR] - Error during installation: System.ArgumentOutOfRangeException: 在多字节的目标代码页中，没有此 Unicode 字符可以映射到的字符。 (异常来自 HRESULT:0x80070459)
   在 System.Windows.Forms.Control.MarshaledInvoke(Control caller, Delegate method, Object[] args, Boolean synchronous)
   在 System.Windows.Forms.Control.Invoke(Delegate method, Object[] args)
   在 ScpDriver.ScpForm.<>c__DisplayClass6.<btnInstall_Click>b__2() 位置 d:\Development\C#\ScpServer\Source\ScpInstaller\ScpForm.cs:行号 244
2015/9/21 23:43:34 [INFO] - Install Succeeded.
2015/9/21 23:43:34 [INFO] - -- Install Summary --

nefarius, 2015-10-01:
Errr... I can't read that =/

sxd1140, 2015-10-02:
i don't know about c#, just check ScpForm.cs  line 244 plz

r15ch13, 2015-10-02:
Google Translate :D
```
2015/9/21 23:43:34 [ERROR] - Error during installation: System.ArgumentOutOfRangeException: In the target multi-byte code page, you do not have this Unicode character can be mapped to the characters. (Exception from HRESULT: 0x80070459)
In System.Windows.Forms.Control.MarshaledInvoke(Control caller, Delegate method, Object[] args, Boolean synchronous)
In System.Windows.Forms.Control.Invoke(Delegate method, Object[] args)
In ScpDriver.ScpForm.<>c__DisplayClass6.b__2() Location d:\Development\C#\ScpServer\Source\ScpInstaller\ScpForm.cs:Line number 244
```

nefarius, 2015-10-02:
The error occurred in an outdated project, please try the latest build!

sxd1140, 2015-10-04:
v1.5.5753.28808 same error
but different stack
new GUI can't copy log

at WdiWrapper.cs:163
at DriverInstaller.cs:24
at DriverInstaller.cs:23
at MainWindow.xaml.cs:325

nefarius, 2015-10-04:
That's why there are `*.log.xml` files generated you can submit, I don't want someone to just copy the GUIs output since there's a lot of information missing.

sxd1140, 2015-10-04:
can't attach files here....
https://gist.github.com/sxd1140/6becc6dae0cc5b5826bd

sxd1140, 2015-11-18:
i reinstall v1.6.126.15319. no more error. great!
I can connect PS3 controller via usb. thanks.
but not work via bluetooth.

`USB\VID_8087&PID_07DC` look like not support yet?

thanks all, great job!

nefarius, 2015-11-29:
Since the original issue seems fixed I'm closing this one.

closed 2015-11-29 by nefarius