nefarius/ScpToolkit #14: Selecting dongle

blubbsy, 2015-09-21:
hi,

is it somehow possible to choice on which bluetooth dongle i want the drivers to be installed? or can you confirm that there is no problem with other bluetooth devices at the same time?

i have the built in bluetooth and a bluetooth dongle, both worked with the previous versions (before you forked it). But i always installed the driver for the dongle and i used the bluetooth headset with the internal.

nefarius, 2015-10-01:
I will add some sort of blacklist or selection screen to the driver installer to avoid this. By default it just replaces the driver for all devices found in `ScpControl.ini`attached to your system.

nefarius, 2015-11-29:
The latest version 1.6.x driver installer has been re-designed to let the user chose which devices shall be installed.

closed 2015-11-29 by nefarius