nefarius/ScpToolkit #15: Error during driver installation

AeonLucid, 2015-09-22:
Hi,

I got myself an error in the driver installation.
```Error during installation: System.IO.FileLoadException: Could not load file or assembly 'file:///C:\Programs\ScpServerNew\ScpService.exe' or one of its dependencies. Operation is not supported. (Exception from HRESULT: 0x80131515)
File name: 'file:///C:\Programs\ScpServerNew\ScpService.exe' ---> System.NotSupportedException: An attempt was made to load an assembly from a network location which would have caused the assembly to be sandboxed in previous versions of the .NET Framework. This release of the .NET Framework does not enable CAS policy by default, so this load may be dangerous. If this load is not intended to sandbox the assembly, please enable the loadFromRemoteSources switch. See http://go.microsoft.com/fwlink/?LinkId=155569 for more information.
   at System.Reflection.RuntimeAssembly._nLoad(AssemblyName fileName, String codeBase, Evidence assemblySecurity, RuntimeAssembly locationHint, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   at System.Reflection.RuntimeAssembly.InternalLoadAssemblyName(AssemblyName assemblyRef, Evidence assemblySecurity, RuntimeAssembly reqAssembly, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   at System.Reflection.RuntimeAssembly.InternalLoadAssemblyName(AssemblyName assemblyRef, Evidence assemblySecurity, RuntimeAssembly reqAssembly, StackCrawlMark& stackMark, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   at System.Reflection.RuntimeAssembly.InternalLoadFrom(String assemblyFile, Evidence securityEvidence, Byte[] hashValue, AssemblyHashAlgorithm hashAlgorithm, Boolean forIntrospection, Boolean suppressSecurityChecks, StackCrawlMark& stackMark)
   at System.Reflection.Assembly.LoadFrom(String assemblyFile)
   at System.Configuration.Install.AssemblyInstaller.set_Path(String value)
   at System.Configuration.Install.AssemblyInstaller..ctor(String fileName, String[] commandLine)
   at ScpDriver.ScpForm.<>c__DisplayClass6.<btnInstall_Click>b__2() in d:\Development\C#\ScpServer\Source\ScpInstaller\ScpForm.cs:line 269```

Sorry, there was no log file created.
I am on Windows 7, already rebooted my PC and tried again.

Bluetooth dongle:
USB\VID_0DF6&PID_2200&REV_1958
USB\VID_0DF6&PID_2200

PS3 Controller:
USB\VID_054C&PID_0268&REV_0100
USB\VID_054C&PID_0268

Both are under 'libusbK USB Devices' in device manager.
The latest version by ScarletCrush worked fine.

The other issue is when I open ScpServer.exe.
![screenshot_1](https://cloud.githubusercontent.com/assets/4643257/10032428/59f5496e-6183-11e5-8d41-42f3f6a537a3.png)

This is how ScpServer looks. (No ScpService is running in the background)
![screenshot_2](https://cloud.githubusercontent.com/assets/4643257/10032449/74e6a4e8-6183-11e5-9e57-0952889889ab.png)

Any fix?

AeonLucid, 2015-09-22:
I fixed it by running the following command in PowerShell 4.0.
````Get-ChildItem -Path 'C:\Programs\ScpServerNew\' -Recurse | Unblock-File```

However, I have an invisible PS3 controller connected.
[image]

AeonLucid, 2015-09-22:
I also have two xbox 360 drivers somehow, tried uninstalling them both, but they came back (both) after I turned on my PS3 controller.
[image]

nefarius, 2015-12-12:
Is there still an issue?

nefarius, 2016-04-02:
Closed. Reason: no feedback.

closed 2016-04-02 by nefarius