nefarius/ScpToolkit #19: Unable to install driver

PFCKrutonium, 2015-09-26:
9/25/2015 8:40:05 PM [INFO] - SCP Driver Installer 1.5.5740.18577 [09/25/2015 20:40:05]
9/25/2015 8:40:08 PM [DEBUG] - Preparing to load DIFxAPI
9/25/2015 8:40:09 PM [INFO] - Running as 64-Bit process
9/25/2015 8:40:09 PM [DEBUG] - DIFxAPI path: C:\SCP\bin\DIFxApi\amd64\DIFxAPI.dll
9/25/2015 8:40:09 PM [DEBUG] - Loaded library: C:\SCP\bin\DIFxApi\amd64\DIFxAPI.dll
9/25/2015 8:40:11 PM [INFO] - Microsoft Windows 10 Home Insider Preview (6.2.9200.0 AMD64) detected
9/25/2015 8:40:11 PM [INFO] - Selected Win10 configuration
9/25/2015 8:40:21 PM [INFO] - ENTER:  DriverPackageInstallW
9/25/2015 8:40:23 PM [INFO] - Looking for Model Section [SCProductions.NTamd64]...
9/25/2015 8:40:23 PM [INFO] - Installing INF file 'C:\SCP\bin\System\ScpVBus.inf' (Plug and Play).
9/25/2015 8:40:23 PM [INFO] - Looking for Model Section [SCProductions.NTamd64]...
9/25/2015 8:40:23 PM [INFO] - Installing devices with Id "root\ScpVBus" using INF "C:\WINDOWS\System32\DriverStore\FileRepository\scpvbus.inf_amd64_21ca245c0e23d7a5\scpvbus.inf".
9/25/2015 8:40:23 PM [INFO] - Will force install because driver is not better and force flag is set.
9/25/2015 8:40:23 PM [INFO] - ENTER UpdateDriverForPlugAndPlayDevices...
9/25/2015 8:40:25 PM [INFO] - RETURN UpdateDriverForPlugAndPlayDevices.
9/25/2015 8:40:25 PM [INFO] - Installation was successful.
9/25/2015 8:40:25 PM [INFO] - Install completed
9/25/2015 8:40:25 PM [INFO] - RETURN: DriverPackageInstallW  (0x0)
9/25/2015 8:40:26 PM [DEBUG] - Preparing to load libwdi
9/25/2015 8:40:26 PM [INFO] - Running as 64-Bit process
9/25/2015 8:40:26 PM [DEBUG] - libwdi path: C:\SCP\bin\libwdi\amd64\libwdi.dll
9/25/2015 8:40:27 PM [DEBUG] - Loaded library: C:\SCP\bin\libwdi\amd64\libwdi.dll
9/25/2015 8:40:28 PM [INFO] - Device with specified VID (8087) and PID (07DC) found, preparing driver installation...
9/25/2015 8:40:35 PM [INFO] - Driver "BthDongle_ef684c28-cd20-472f-8d2a-8c8a8af76427.inf" successfully created in directory "Driver"
9/25/2015 8:40:40 PM [INFO] - Installation result: WDI_SUCCESS
9/25/2015 8:40:40 PM [INFO] - Installed driver for Bluetooth dongle USB\VID_8087&PID_07DC
9/25/2015 8:40:40 PM [ERROR] - Error during installation: System.IO.FileLoadException: Could not load file or assembly 'file:///C:\SCP\bin\ScpService.exe' or one of its dependencies. Operation is not supported. (Exception from HRESULT: 0x80131515)
File name: 'file:///C:\SCP\bin\ScpService.exe' ---> System.NotSupportedException: An attempt was made to load an assembly from a network location which would have caused the assembly to be sandboxed in previous versions of the .NET Framework. This release of the .NET Framework does not enable CAS policy by default, so this load may be dangerous. If this load is not intended to sandbox the assembly, please enable the loadFromRemoteSources switch. See http://go.microsoft.com/fwlink/?LinkId=155569 for more information.
   at System.Reflection.RuntimeAssembly._nLoad(AssemblyName fileName, String codeBase, Evidence assemblySecurity, RuntimeAssembly locationHint, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   at System.Reflection.RuntimeAssembly.InternalLoadAssemblyName(AssemblyName assemblyRef, Evidence assemblySecurity, RuntimeAssembly reqAssembly, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   at System.Reflection.RuntimeAssembly.InternalLoadFrom(String assemblyFile, Evidence securityEvidence, Byte[] hashValue, AssemblyHashAlgorithm hashAlgorithm, Boolean forIntrospection, Boolean suppressSecurityChecks, StackCrawlMark& stackMark)
   at System.Reflection.Assembly.LoadFrom(String assemblyFile)
   at System.Configuration.Install.AssemblyInstaller.set_Path(String value)
   at System.Configuration.Install.AssemblyInstaller..ctor(String fileName, String[] commandLine)
   at ScpDriver.ScpForm.<>c__DisplayClass6.<btnInstall_Click>b__2() in d:\Development\C#\ScpServer\Source\ScpInstaller\ScpForm.cs:line 269

9/25/2015 8:40:40 PM [INFO] - Install Succeeded.
9/25/2015 8:40:41 PM [INFO] - -- Install Summary --
9/25/2015 8:40:41 PM [INFO] - Bus Driver installed
9/25/2015 8:40:41 PM [INFO] - Bluetooth Driver installed

PFCKrutonium, 2015-09-26:
9/25/2015 8:46:26 PM [ERROR] - Error during uninstallation: System.IO.FileLoadException: Could not load file or assembly 'file:///C:\SCP\bin\ScpService.exe' or one of its dependencies. Operation is not supported. (Exception from HRESULT: 0x80131515)
File name: 'file:///C:\SCP\bin\ScpService.exe' ---> System.NotSupportedException: An attempt was made to load an assembly from a network location which would have caused the assembly to be sandboxed in previous versions of the .NET Framework. This release of the .NET Framework does not enable CAS policy by default, so this load may be dangerous. If this load is not intended to sandbox the assembly, please enable the loadFromRemoteSources switch. See http://go.microsoft.com/fwlink/?LinkId=155569 for more information.
   at System.Reflection.RuntimeAssembly._nLoad(AssemblyName fileName, String codeBase, Evidence assemblySecurity, RuntimeAssembly locationHint, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   at System.Reflection.RuntimeAssembly.InternalLoadAssemblyName(AssemblyName assemblyRef, Evidence assemblySecurity, RuntimeAssembly reqAssembly, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   at System.Reflection.RuntimeAssembly.InternalLoadFrom(String assemblyFile, Evidence securityEvidence, Byte[] hashValue, AssemblyHashAlgorithm hashAlgorithm, Boolean forIntrospection, Boolean suppressSecurityChecks, StackCrawlMark& stackMark)
   at System.Reflection.Assembly.LoadFrom(String assemblyFile)
   at System.Configuration.Install.AssemblyInstaller.set_Path(String value)
   at System.Configuration.Install.AssemblyInstaller..ctor(String fileName, String[] commandLine)
   at ScpDriver.ScpForm.<btnUninstall_Click>b__10() in d:\Development\C#\ScpServer\Source\ScpInstaller\ScpForm.cs:line 375

9/25/2015 8:46:26 PM [INFO] - Uninstall Succeeded.
9/25/2015 8:46:26 PM [INFO] - -- Uninstall Summary --

nefarius, 2015-09-28:
You didn't unblock the archive.

closed 2015-09-28 by nefarius