nefarius/ScpToolkit #22: Random button presses

qbitweb, 2015-09-29:
Hello, I just installed ScpServer for my DS3 controller on a Windows 10 x64 machine and while I had no problem whatsoever with the setup I'm getting random button presses on the emulated Xbox 360 controller.

This happens both with USB cable and via Bluetooth, it's like if the pad is sending button presses all the time (only dpad and the four symbol buttons, square, circle, etc).

What's odd is that if I open the SCP profile manager it doesn't seem to receive any of those presses. If I open controller properties from the windows device manager or any game that uses an Xbox pad I get them.

TheVirtuousJ, 2015-11-05:
I am having the same issue. I am currently playing rocket league using the controller. While the RT is down, I am using left stick to turn and O to boost. Sometimes in my excitement I also depress the L3. When this happens, it makes a d-pad press and an X button press. It seems like a ghosting issue or something. I'm not sure if it hits all the buttons or just those 2.

Hyo8745, 2015-11-05:
Guys, try these two things:
1. Press down the D-Pad down/right buttons or L3. Put a little strength. If the controller goes wild, it's hardware problem.
2. Put the USB power cable on the controller and mess a little with the USB contact plug inside the controller (you know, push it a little to the left, right, up, down, but don't remove the plug), if the controller goes wild when you do this, it's the same hardware problem as before.

The solution I found and tested with success on both my controllers is on this video here:

qbitweb, 2015-11-13:
Omg Hyo8745 you're awesome, I followed your tutorial and now the controller works perfectly!

I guess we can close this as it all seems just an hardware problem!

closed 2015-11-13 by qbitweb