nefarius/ScpToolkit #25: Support Macbook Pro 2015 Bluetooth

deegital, 2015-10-07:
Hi, can you add support for the Macbook Pro 2015 Models Bluetooth Chip?
USB\VID_05AC&PID_8290

nefarius, 2015-12-12:
Do you mean you tested it with this chip and I should add it to the Wiki or what du you mean by "add support"?

nefarius, 2016-04-02:
No feedback received.

closed 2016-04-02 by nefarius

MrRinoy, 2016-05-03:
What deegital was trying to say is that the newest 2015 model of the Macbook Pro has an unsupported bluetooth chip. According to the wiki, Apple Broadcom Built-in Bluetooth (MacBookPro8), (MacBookPro9), and (MacBookPro10) are supported. However, the 2015 USB\VID_05AC&PID_8290 chip on the (MacBookPro11) has no support and currently does NOT work. As someone with this model of Macbook Pro who could greatly benefit from an available USB port and doesn't have the technical know-how to do it myself, I'd greatly appreciate it if you found a way to support this chip. If you have any more questions about the chip or need any screenshots, I'd be happy to provide them.