nefarius/ScpToolkit #29: BSOD when driver verify is enabled

mirhl, 2015-10-13:
Tell [verifier](https://support.microsoft.com/en-us/kb/244617) to monitor the driver and you won't be able to boot again in Windows (or at least in my 7x64).

aLagoG, 2015-12-16:
I've been having the same problem.
This is the message that I get after memory dump analizis.

    DRIVER_VERIFIER_IOMANAGER_VIOLATION (c9)
    The IO manager has caught a misbehaving driver.
    Arguments:
    Arg1: 000000000000021f, A driver has not filled out a dispatch routine for a required IRP major function.
    Arg2: fffff800e247b174, The address in the driver's code where the error was detected.
    Arg3: ffffcf8147d54dc0, IRP address.
    Arg4: 0000000000000000

With faulting ip: `ScpVBus+b174`

Hope this helps you find the problem

nefarius, 2016-02-06:
Well, don't run the verifier then :P I don't have a fix for this yet.

nefarius, 2016-03-20:
Ok somehow my run of the verifier doesn't create a dump file >_<

mirh, 2016-03-20:
[?](https://www.sophos.com/en-us/support/knowledgebase/111474.aspx)

nefarius, 2016-03-20:
@mirh yeah, I know right =P But no matter what switch I flip; none of my virtual machines (Win7/8/10) create a f***in' memory dump! I sincerely don't know what I'm doing wrong...

nefarius, 2016-03-20:
Trying one more time with `Complete memory dump` enabled...

mirh, 2016-03-20:
**[?](http://support.kaspersky.com/3054)**

nefarius, 2016-03-21:
**I. FIXED. IT.** !

[image]

charlesmilette, 2016-03-21:
[image]

closed 2016-04-10 by nefarius