nefarius/ScpToolkit #32: DS4 not working via USB

Linus9000, 2015-10-17:
Hay,

I have the problem on Win 10 x64 that the DS4 Controller, when connected via USB, won't be recognized by the SCPServer (neither ScpServer nor ScpService).. I know it worked "sometimes", for example in ScpServer_v1.5.5749.18036, but also worked for some time in ScpServer_v1.5.5756.37820. Bluetooth works fine, but I can't get it to work over USB anymore :-( The DS4 will be recognized by Windows itself, but the buttons are messed up then. When it was connected over Bluetooth when plugging in the USB Cable, it works as two controllers, meaning it issues for example the Arrow Right command twice. This is quite annoying, do you have any suggestions?

nefarius, 2015-10-18:
Look for "Xbox 360 Controller" entries in your device manager and uninstall all of them. Sometimes the virtual Xbox 360 devices get the wrong driver; manually removing them and letting Windows re-discover it with correct drivers helps.

Linus9000, 2015-10-19:
While that, unfortunately, didn't fix the problem, I did a system restore which fixed it. God knows which driver's gotten fragged by the beautiful "Driver Booster", which I subsequently uninstalled. I should've known better.. Sorry for wasting your time with that and thank you very much for taking over this great project :-)

closed 2015-10-19 by Linus9000