nefarius/ScpToolkit #34: I ran CleanWipe and now Bluetooth Refuses to work.

Taconut, 2015-10-20:
All traces of bluetooth are now gone. I can't access the bluetooth settings, add a device, or see the Tray icon. I've already tried rebooting quite a few times and re-enabling the Bluetooth Support Service and Bluetooth Handsfree Service. What should I do?

Taconut, 2015-10-21:
Alright, so here's my updated issue. I can transfer it to another issue if you'd like.

When I install the bluetooth driver, all bluetooth functionality disappears from my computer. I don't know what I should do to help solve the issue, but I *can* say which version the issue started in. Version v1.3.5689.40377 works perfectly, however v1.4.5706.37073 has the same issue. The problem must have arisen somewhere between those two versions.

nefarius, 2015-11-29:
If you want to use your controllers wireless you **must** dedicate one Bluetooth dongle to the software. This is why the icon disappears from the tray; you **can not use any other Bluetooth devices** from the same dongle after the driver got changed.

closed 2015-11-29 by nefarius