nefarius/ScpToolkit #36: Issue with service starting

Tythesly, 2015-10-21:
Im trying to get this program working on my Windows Vista 64bit Virtual Machine so i can play assassins creed on steam

Everything runs and seems like it will work but the computer never recognizes it as a Playstation 3 controller... in game controllers it shows as Controller (XBOX 360 For Windows) and when i click on properties non of the buttons work.

I noticed that the SCP DSx Service wasnt started and when I try to manually start it I get this message after 2 minutes. - http://prntscr.com/8tux0f

http://prntscr.com/8tuyvw - Hardware IDS

This is the first time i ran this new projects version. On my windows 7 computer the last stable build of the original program works great.

if you need more info let me know

nefarius, 2015-11-29:
Is this still an issue with the latest version?

Tythesly, 2015-12-01:
Honestly dont know... I thought nobody would reply lol... I started using the Old scarlet productions program and it worked.... I guess ill reinstall your project and will let you know what happens. Give me a couple of days.

Tythesly, 2015-12-21:
So I lost my VM when my SSD drive crashed... and they want me to pay $20 for a new Vista product key. If I try again and it doesnt work ill reopen a new ticket

nefarius, 2015-12-21:
Well, Backups... ;P

Tythesly, 2015-12-21:
Well at least I can report your latest version works great on my windows 7
professional 64 bit :)

closed 2018-01-26 by nefarius