nefarius/ScpToolkit #39: Rumble is causing the controller to freeze for ~10 seconds

7yphoid, 2015-10-28:
I have installed ScpServer on Windows 10 64-bit. I am using a wired DS3 controller.

It seems to be working fine except for the fact that whenever the controller rumbles (let's say I get hit in the game I'm playing), the controller freezes and the input "sticks" for ~10 seconds. So, if I was walking right when it happened, I'll continue walking right until it unfreezes.

Interestingly, while disabling rumble in ScpSettings does make the controller not rumble, it does not prevent the controller from freezing after rumble events.

Any clue what the cause of this might be and what I can do to remedy this problem?

CookiePLMonster, 2015-11-03:
Come to think of it, my issue might be similar - rumble seems to send some random data on analog sticks, effectively locking them.

Do note this is for wired. Would still like to debug it and try to come up with a solution, though for the second time I don't remember where the input gets processed. Meh.

7yphoid, 2015-11-03:
For me, the rumble actually causes the whole controller to lock up. I can't even pause the game until the ~10 seconds is up. Also, IIRC, if I hit the pause button on the controller, it does pause after it's done locking up, which means that the buffer is still active but not being processed.

How should I go about gathering additional debugging information? What should I be looking at?

I'm using MotionInJoy until I can get SCP to work. At least I'll be getting a Steam Controller in a week.

felixthecat1970, 2015-11-28:
same here windows 10 pro TH2, rumble stop control funtions, test release ScpToolkit v1.6.133.15324 and ScpToolkit v1.6.126.15319; but release ScpServer v1.5.5756.37820 works ok with rumble.

1st8, 2015-11-28:
Experiencing the same, will try v1.5.

RyanTH98, 2015-12-02:
Confirmed on the latest version. Tried playing Rocket League, experience lockups, and weird rumbling.

skite2001, 2015-12-03:
got the same problem. worked the whole time, after updating to latest drivers the problem happens. for example: pressing shoot button -> whole mag gets fired, after that the rumble kicks in. after the rumble is over i can move my guy again.

http://forums.pcsx2.net/attachment.php?aid=58290
ScpToolkit_Log-Package_skite2001_1449181443.zip

nekogami, 2015-12-10:
Just had the same problem, installing ScpToolkit v1.6.186.15342-BETA fixed it.

skite2001, 2015-12-10:
can confirm: latest beta fixes the rumble issue

nekogami, 2015-12-11:
Had a little problem last night with v1.6.186.15342-BETA .
On Rocket league, some buttons were pushed while playing without any action which made me jump/loose the ball cam randomly.
Had to rollback to https://github.com/nefarius/ScpServer/releases/tag/v1.5.5787.40106

ChaoticTabris, 2016-01-09:
Can confirm this problem or a similar one is still happening to me on v1.6.229.15365. It freezes for a moment and i hear the sound as if a USB device had been unplugged. At times i also had the control controller completely stop working and being "stuck" vibrating until i unplugged it (i don't have a device to use it with Bluetooth, so i only play with the USB cable).

closed 2018-01-26 by nefarius