nefarius/ScpToolkit #41: Low Battery OSD

Blackbird88, 2015-10-31:
Something like ths would be very useful!
[image]

nefarius, 2015-12-12:
Right, I have some prototypes in the works, stay tuned!

Blackbird88, 2015-12-12:
Cool!

nefarius, 2015-12-13:
What do you think?

[image]

Blackbird88, 2015-12-13:
Very good. You should make it configurable so you can set the threshold when it shows up (and maybe even change colors to yellow and then red when it's low?). Kinda like battery LED on phones.
Optional one-time blinking would be perfect!

teeedubb, 2015-12-31:
It is great that to see that this feature has interest.

Any chance that a low battery notification sound be added too?

Jan200101, 2016-01-01:
hmmm.... maybe adding a sound to it like a few silent beep

FuzzyFoxe, 2017-08-17:
It's been a couple years, any progress on this? o:

nefarius, 2017-08-19:
[Well, more or less](https://github.com/nefarius/Indicium-Supra) 😄

closed 2018-01-26 by nefarius