nefarius/ScpToolkit #45: Change Controller slot

sueds, 2015-11-10:
Hi, 

On my computer I'm already using a gamepad and I do wish to add a ps3  gamepad but just for the second player. I guess I have to change the scp monitor project but I don't find any clue on how to do it. If could point me in the right direction that would be cool.

nefarius, 2015-11-29:
This is currently not possible.

mirhl, 2015-12-04:
You can adjust controllers order [with x360ce](http://pcgamingwiki.com/wiki/Xbox_360_Controller_Emulator#Unset_controller_from_first_player). 
And I'm not sure this is something better to be done on the driver level.

nefarius, 2016-04-02:
This will be possible in the next release.

mirh, 2016-04-02:
[XOutput](https://github.com/nefarius/ScpToolkit/commit/85cc28dd30ff9489da047b82f4b8cd7c743c307c) magic right?
Though, I still wonder: isn't controller slot something "system"-wide rather than "controllers"-specific?
Why changing this should be *conceptually* in the driver?

nefarius, 2016-04-02:
It isn't and won't be possible to inflience which XInput slot the controller gets assigned because that's dictated by Microsofts `xusb.sys` driver alone. The bugyness with mixed controller environments will be solved though, that's why I commented on this issue.

mirh, 2016-04-02:
Oh cool. 
Could that even help with #145 and #146 ?

nefarius, 2016-04-02:
@mirh #145 is affected and fixed in the upcoming release, I'm still not sure what i can do about #146 though.

nefarius, 2016-04-02:
Finally making use of those milestones :D

charlesmilette, 2016-04-02:
Do you have an estimate for the release date of 1.7.x?

nefarius, 2016-04-02:
@charlesmilette since this weekend got pretty much occupied with other stuff, I can release a beta hopefully within the following week or the upcoming weekend.

nileplumb, 2016-04-03:
@nefarius Looking forward to it, have been checking the github commits for vbus and toolkit daily haha getting close it seems! Will be able to report issues in a 4 player environment on occasion too :)

closed 2016-04-10 by nefarius

pnutus, 2016-06-18:
This issue is closed, but I can't find any info on how to change controller slots. Am I missing something?

mirh, 2016-06-18:
As I said. Use x360ce.

Zikasuo, 2017-01-20:
So My issue is kind of similar i have one controller and one keyboard and mouse. i want to use my keyboard and mouse for the player 1 slot and the controller as player 2. My controller is a ps3 dual shock 3 controller, and i only have one controller available. i have emulated it through x360ce and using SCP... but i'm only able to achieve the ability to control both on the controller. if i'm doing something wrong or there's an easier way to do this please let me know. thanks!

mirh, 2017-01-20:
http://pcgamingwiki.com/wiki/Xbox_360_Controller_Emulator#Change_controller_slot

Zikasuo, 2017-01-20:
I did that but since x360ce doesn't run my PS3 controller by itself I need SCP to detect it and then x360ce to map the controller

mirh, 2017-01-20:
> I need SCP to detect it and then x360ce to map the controller

Exactly, so please stop commenting the driver.

Zikasuo, 2017-01-20:
? Am I doin something wrong