nefarius/ScpToolkit #47: DS4 audio passthrough

kevmegforest, 2015-11-14:
Get bluetooth audio working

CookiePLMonster, 2015-11-18:
This issue sounds more like a demand, thus do you have any tips on how to do it? As far as I'm aware, the protocol hasn't been reversed by anyone at that time yet.

nefarius, 2015-11-29:
I have no idea how the audio transmission is accomplished and I' not aware of anyone having reverse engineered it yet. Maybe Sony uses A2DP for it's audio communication but I haven't looked into it yet and doubt that I will anytime soon.

closed 2015-12-15 by nefarius

mirhl, 2015-12-28:
Audio is [only](http://eleccelerator.com/wiki/index.php?title=DualShock_4#Bluetooth) handled in PS4 mode (which means already paired?), but it dis[appear](http://gaming.stackexchange.com/questions/188314/bluetooth-audio-streaming-from-mac-pc-to-dualshock-4) soon because it [expects](http://forums.pcsx2.net/Thread-DS4Windows-yet-another-DualShock-4-driver-No-longer-under-active-development?page=8#pid364459) [to be already paired](http://forums.pcsx2.net/Thread-DS4-To-XInput-Wrapper?page=203) in the first place. 

Assumed this problem was overcome (ideally requires a real ps4 to spoof), sound should shouldn't be hard to get through normal [headset profile](https://gist.github.com/johndrinkwater/7708901#file-dualshock-research-L109). 
Since [HSP](https://en.wikipedia.org/wiki/List_of_Bluetooth_profiles#Headset_Profile_.28HSP.29) reportedly only supports CVSD (monophonic and just crappy for general audio) and PCM (too much bandwidth intensive afaik) though, I wonder if [A2DP](http://www.psdevwiki.com/ps4/DS4-BT#PS4) couldn't still be involved. 

Also, Jays2Kings/DS4Windows#17

dreamcat4, 2016-01-09:
OK
It sounds like someone with real PS4 needs to get a packet dump for us, like with some 3rd party listening in. However bluetooth connection is encrypted. So surely you would also need to dump the encryption key which is negotiated during pairing or at beginning of session. And use some method to decrypt the key to use to decrypt the packet log... sounds pretty tough.

That is not simple like wireshark where you can just listen direct those communication back-forth.

So instead I would suggest maybe attach some hardware probe directly to the BT controller chip on the DS4 pcb... That is not simple either.

Forgive me if I misunderstand. It is just a speculation. Perhaps there is indeed an easier way. Like a debug hardware or some possible debug mode. Or other pre-existing known portable device (e.g. PS Vita or sony bluetooth audio products etc) which may also share similar / known protocol.

mirhl, 2016-01-10:
Everything about [spoofing](http://eleccelerator.com/weekly-report-february-23-2014/) is already [documented](http://eleccelerator.com/keyboard-and-mouse-for-playstation-4-games-second-prototype/#more-487). 

Moreover, with PS4 being able to run Linux, perhaps in a few years special hardware won't even be required.

kq01526, 2016-01-10:
Maybe some more people should ask for it on Twitter:

https://twitter.com/GI_FTW/status/686148538959532032

;)

alkatena, 2017-05-07:
Hi people, I've just been give a DS4 controller and was thinking about using its headphones jack so I came across this thread.

In Amazon, there is an official Sony BT adapter for the DS4. which claim:

> Utilize every feature of DUALSHOCK 4, including buttons, analog sticks, touch pad, light bar, motion sensors, vibration and **stereo headset jack**, as long as the application supports these features.

Has anybody own this BT adapter? Could you please confirm that if I plug a headset, it's going to work?

[https://www.amazon.com/Sony-DualShock-4-USB-Wireless-Adaptor/dp/B01KWLKKQU/ref=sr_1_1?ie=UTF8&qid=1494185466&sr=8-1&keywords=ds4+bluetooth](url)

vpontin, 2017-11-12:
Also for your all information: https://github.com/Ryochan7/DS4Windows/issues/121