nefarius/ScpToolkit #48: SCP DSx Service fails to start automatically

fernandofj, 2015-11-14:
On boot, SCP DSx Service fails. I'm running Windows 10 Pro Build 10240, and tried two workarounds but none worked:

 - Setting the service to restart on first and second failures;
 - Increasing the timeout as described here: https://support.microsoft.com/en-us/kb/922918

Found these event logs that seem related to the issue:

Log Name:      System
Source:        Service Control Manager
Date:          11/14/2015 4:14:41 PM
Event ID:      7000
Task Category: None
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      PC
Description:
The SCP DSx Service service failed to start due to the following error: 
The service did not respond to the start or control request in a timely fashion.

Log Name:      System
Source:        Service Control Manager
Date:          11/14/2015 4:14:41 PM
Event ID:      7009
Task Category: None
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      PC
Description:
A timeout was reached (30000 milliseconds) while waiting for the SCP DSx Service service to connect.

fk33, 2015-12-08:
I had the same problem and was able to work around it by increasing the ServicesPipeTimeout to 120 seconds.

Jan200101, 2015-12-12:
Same Issue but fk33´s method didn´t work for me.

charlesmilette, 2016-02-14:
Same here. Windows 10 Home 64 bits build 10586

charlesmilette, 2016-02-14:
You can simply fire it up manually.

charlesmilette, 2016-02-15:
#215

mirh, 2016-02-15:
@charlesmilette 
To be honest this is no W10 specific. I had similar issues with the service even on 7.

charlesmilette, 2016-02-16:
Ha. Okay. All people who reported Windows' version said Windows 10, so I tought it might be because of that  
I think the problem is, the service starts up while everything else is also starting, that actually slowdowns the service's startup over the timeout  
This might help. You can probably create a little program that waits until the processor usage goes under 10% or something and starts the service after that. Since the other things would have finished startup, the service will start in a reasonnable time

These are all assumptions, nothing actually real.

mirh, 2016-02-16:
That should be already handled by "delayed start" setting. 
Though, I think logic shouldn't be put towards that. 

But rather in don't making windows kill the hanging process (ie: by avoiding race conditions)

charlesmilette, 2016-02-16:
Probably [this](http://stackoverflow.com/questions/216401/windows-service-startup-timeout#216487) can help. If I read correctly, sending SetServiceStatus will prevent any timeout.

charlesmilette, 2016-02-16:
I just started my computer today, and left it on the lockscreen for at least 10 minutes. After opening my session, I noticed the service started correctly. Probably that opening a session is the problem that delays the service enough for timeout.

closed 2018-01-26 by nefarius