nefarius/ScpToolkit #50: PS3 Wireless Controller constantly flashes

Dodgexander, 2015-11-19:
I have a standard (not dual shock) ps3 controller and when I connect the device via cable the LED display works fine.

When I disconnect and try to connect via Bluetooth though the LED's don't work properly anymore. They constantly flash as if the device hasn't connected or is low on battery.

At this point the controller works fine, but continues to flash like it isn't connected or low on battery. I have made sure its fully charged and that its connected according to the scpmonitor.

I wonder what has changed because with the older versions of the application the LED indicators worked fine via Bluetooth.

AlbinoDrought, 2015-11-19:
Happens to me as well, attached log files renamed to .txt

[system.txt](https://github.com/nefarius/ScpServer/files/39583/system.txt)

[ScpDsxService_DESKTOP-OR18CP6$.log.txt](https://github.com/nefarius/ScpServer/files/39584/ScpDsxService_DESKTOP-OR18CP6.log.txt)

[ScpDsxMonitor_Sean.log.txt](https://github.com/nefarius/ScpServer/files/39585/ScpDsxMonitor_Sean.log.txt)

[ScpDsxInstaller_Sean.log.txt](https://github.com/nefarius/ScpServer/files/39586/ScpDsxInstaller_Sean.log.txt)

[HID-Devices.txt](https://github.com/nefarius/ScpServer/files/39587/HID-Devices.txt)

[USB-Devices.txt](https://github.com/nefarius/ScpServer/files/39588/USB-Devices.txt)

[setupapi.dev.log.txt](https://github.com/nefarius/ScpServer/files/39589/setupapi.dev.log.txt)

logicwrath, 2015-11-20:
Happens to me now with many different versions.  It must be something on Windows 10.  Someone on the Internet blamed TH2 update, I have not tried removing it to test.  I would love to see this resolved.

I can get it partly working with the USB cable, but it flashes.  I have to constantly unplug and re-insert the USB cable because the controller will crash or twitch repeatedly every so often

Dodgexander, 2015-11-20:
I also have the latest update on Windows 10.

For me the functionality works fine. The controller indicators just don't work over Bluetooth.

Even when they are flashing on the controller the controller still works fine and you see the Bluetooth device fine in device manager. You even  hear the connect/disconnect sounds.

As for the command error message at start-up, I don't know if that's to blame but it's just annoying each time you're computer starts.

AlbinoDrought, 2015-11-20:
Just compiled the latest version and everything is working fine over bluetooth. (18572bc8d024556805f051d9562286f154d0c43b)

Also enabled and disabled "Flash LED when charging" for the "Pad ID" option, although I doubt this was it.

Dodgexander, 2015-11-20:
Thats great if its been fixed, but how can we get the latest version without compiling?

Dodgexander, 2015-11-20:
Ps this is the error on startup for me [img]http://i.imgur.com/YfbvSTu.png[/img]

Dodgexander, 2015-11-21:
Thanks for fixing this in the latest release! Is there anything I can do about the "invalid command line" error message on start up? Should I open a separate issue?

nefarius, 2015-11-21:
I used CCleaner to pin down the startup entry that's causing the error message and deleted it. Sadly didn't make a screenshot so I can't recall the exact name of the registry key...

closed 2018-01-26 by nefarius