nefarius/ScpToolkit #51: Bluetooth not working after restart.

TheHardew, 2015-11-23:
After drivers installation I can play with my ds3 via bluetooth, HCI version in scp monitor is 6.xxx. If I restart my computer it is 5.xxx and I cannot connect the pad with bt, but it still works fine with wire. To connect it again I have to uninstall drivers (I do it manually) and install them again.

TheHardew, 2015-11-30:
Now I see it also happens if I suspend my pc.

TheHardew, 2015-12-01:
When I set led status in scptoolkit hci again revrted back from hci 6.0102 to 5.0102

theycallmeshua, 2015-12-07:
ok, so i had this same problem and after a few days heres what finally got it to work for me(win 10 x64).

1. navigate to C:\Program Files\Nefarius Software Solutions\ScpToolkit , right click>properties>compatibility> run as administrator check box  , for each of the following 
 -scpdriverinstaller
 -scpmonitor
 -scppair
 -scpservice
 -scpsettings

2. with controller connected via usb open device manager. under libusbk usb devices 
 - right click>uninstall> delete driver software for device.> ok , for
    --bluetooth module
    --playstation controller

DO NOT RESTART YET 

3. run scp toolkit settings manager
 -under service settings unselect all options

RESTART

4. open device manager and make sure the original drivers installed for bluetooth and controller

5.
   a. if no. uninstall drivers again until you get original drivers
   b. if yes. run scptoolkit driver installer. 
 -select server, Bluetooth, controller and then install

6. run scp toolkit settings manager
 -under service settings select bluetooth driver

your controller should currently connect via bluetooth and recontect after restart. this worked for me, hopefully it worked for you. it was a terrible work around but it did work

theycallmeshua, 2015-12-08:
spoke too soon, worked for a few hours and several restarts but eventually stopped working again

Jan200101, 2015-12-12:
same for me but even worst. I wont detect the Computer at all not even wired.
Like it forgot how to connect to it but the drivers are all still working.... I think its something with ScpService because it worked before i restarted (It was on) and it wasnt after I restarted... putting it onto the Autostart list wont fix the issue

theycallmeshua, 2015-12-14:
It's definitely an issue with the software as the the Bluetooth card is still picked up by the computer but the software no longer recognizes it appropriately.  I have debug files from everything working and then another set from after a system restart when everything stops working.  Will post them when my tablet is done charging in case they can be of some help.

theycallmeshua, 2015-12-17:
after install where everything works fine
[ScpToolkit_Log-Package_Shua_1449689623_wiredBTworking.zip](https://github.com/nefarius/ScpServer/files/65902/ScpToolkit_Log-Package_Shua_1449689623_wiredBTworking.zip)

after restart where bt stops connecting but wired still works fine
[ScpToolkit_Log-Package_Shua_1449690002_after_restarBTnot_working.zip](https://github.com/nefarius/ScpServer/files/65903/ScpToolkit_Log-Package_Shua_1449690002_after_restarBTnot_working.zip)

charlesmilette, 2016-02-14:
Same here. Genuine Controller.

charlesmilette, 2016-02-14:
But I don't see any regression in HCI version when it stops working.

charlesmilette, 2016-02-15:
#215

erapkin, 2016-02-22:
#224, #104 all seem to be related.  I haven't been able to come up with a permanent solution.

Jan200101, 2016-03-04:
found a fix and uploaded it into #224 you have pick the right comments are mine so look for these they have 1. a tool that checks if Bluetooth should be working 2. The fix

Jan200101, 2016-04-06:
@theycallmeshua you do know that you have to pick the Bluetooth donge or card in the driver installer so its either a faulty dongle / card or you just have another issue

theycallmeshua, 2016-04-06:
@jan200101 first.  Why did you mention me for no apparent reason, when my last post was four months ago.   second why did you feel the need to be condescending.  Third of course I know that,  if you read any of my posts it's quite obvious that I can get the driver to work which wouldn't be the case if I wasn't able to install it.  

That being said,  the card is not faulty.  The normal Bluetooth driver works perfectly fine when installed, the results I got are specific to using the toolkit.  I just got a dongle that I am going to set up eventually but haven't had the time to mess around with it yet. It is quite possible I have another issue as the tablet I'm using is a little dated. I'll figure it out when I have time.

charlesmilette, 2016-04-10:
Any news about this one, @nefarius?

nefarius, 2016-04-10:
@charlesmilette no but seeing it's an integrated (mobile) chipset I doubt I'll ever investigate or attempt to fix it since they typically do some weird sh*t off the specs only the vendor drivers are capable off, especially with power management. You save yourself a lot of troubles if you just use a dedicated stand-alone dongle.

forbjok, 2016-06-12:
I'm having the same issue with a stand-alone BT dongle, both on my gaming laptop and desktop. I have to run the SCP driver installer on the dongle every time I reboot for the controller to connect.

Never had this issue before with the DS3 driver or the old SCP v1.2.2 driver, with the same dongle and machines.

mirh, 2016-07-05:
Try to put SCP service on "Late" start.

Jan200101, 2016-07-12:
Same as above but also set it to restart if it crashes

HowlingFoxRouko, 2016-07-24:
((DS3 ONLY HERE)) I was having the same problem with one of my two controllers. If you can, obtain a second controller and try to pair it to PC the normal way, if it pairs and if your controller (the one that doesn't work with Bluetooth) works when wired into your PC via USB, try hitting the reset button located on the bottom of the controller near the middle screw (the small, needle-sized hole to the immediate right of the controller, you will need something like a paperclip to press it.) This will power off your controller and pressing the PS button will power it back on, and it should be able to pair afterward.

Hope this helps! ^^

P.S. Yeah, I felt stupid after taking the controller apart and realizing the fact afterward as I had isolated it to not being so much a software problem as it was a hardware problem. Don't be like me. XD

tosho, 2016-08-30:
I have Sony DS3 and I'm on Win 10 Pro x64.
The controller is not connecting via Bluetooth after restart of the PC.
First time I used ScpToolkit - few years ago - on Win 8 Pro it was working flawlessly. 
I noticed this problem to occur after I upgraded to Win 10 Pro - actually it was working at the begging but maybe few months later (after some Win updates) it start making problems.

Those workaround doesn't work for me. I uninstall and Install the drivers every time after restart.
[ScpToolkit_Log-Package_Tosho_1471945374.zip](https://github.com/nefarius/ScpToolkit/files/444625/ScpToolkit_Log-Package_Tosho_1471945374.zip)

tosho, 2016-09-04:
following @theycallmeshua instructions commented on 7 Dec 2015 doesn't help..
my DS3 still can't connect via Bluetooth after the  restart. I'm on Win 10x64 Pro now but when I was using Win 8 x64 Pro I didn't had that isssue. I''m using the same Bluetooth adapter which is motherboard integrated - Qualcomm Atheros AR9462 Bluetooth 4.0 + HS Adapter. I also tried the Delay solution - with no success as well.
Any new ideas how to fix this?
[ScpToolkit_Log-Package_Tosho_1471945374.zip](https://github.com/nefarius/ScpToolkit/files/453811/ScpToolkit_Log-Package_Tosho_1471945374.zip)

Freddy-Francesco, 2016-11-13:
Experiencing same issue here. Are there any older versions that don't have this problem?

a1studmuffin, 2017-01-02:
Same issue here, Windows 8.1 w/Qualcomm Atheros AR3012 Bluetooth 4.0 + HS. Did a clean install and was able to connect 4 PS3 controllers over USB and bluetooth. After rebooting they'll connect over USB but not bluetooth.

[ScpToolkit_Log-Package_Michael_1483326311.zip](https://github.com/nefarius/ScpToolkit/files/680336/ScpToolkit_Log-Package_Michael_1483326311.zip)

forbjok, 2017-01-02:
It's already fixed in the latest beta.
The beta also has a ton of other improvements over the latest "stable" release.

a1studmuffin, 2017-01-02:
Awesome, thanks for the quick response!

PippBauda, 2017-01-04:
Hi all. I install the toolkit without problems, in particular the latest beta to prevent the problem about the bluetooth pair but it has not been solved and my HCI change from 6 to 5.0102. I noticed that the HCI doesn't change if i reboot. It only change when i **SHUT DOWN** and start my pc again. Everything is fixed if i rollback to my original drivers, reboot(sistem requirement) and reinstall the bluetooth driver of scp.

Can someone help me? Thanks.

charlesmilette, 2017-01-04:
Try to disable Fast Startup (on mobile right now, can't link. Just Google it.)

PippBauda, 2017-01-04:
@charlesmilette Thanks for the fast answer. Unfortunatly this can't help me cause i've already disabled fast boot. Any other ideas?

charlesmilette, 2017-01-04:
Hmmm... Try to force Bluetooth driver in the settings, and make sure
that the Scp service is running (sometimes it gets killed when it
takes too long to start)

PippBauda, 2017-01-04:
@charlesmilette I checked for "force Bluetooth driver" in settings and it is yet on... I have checked if the service is on and it starts without problems at start up. I read about a possibile solution here in GitHub that consist in set the scp service in "automatic delayed start" but the result is the same: when the service starts after a reboot all works fine. If i shut down, when i boot the pc, after few minutes the service start and the notification tray notify the detection of a Bluetooth device with HCI 5.0102 and VCI 6(same ad before shut down).

moogy6610, 2017-01-17:
i have the same problem with the wiring connection i have to install the drivers every time i restart

mirh, 2017-01-17:
For usb you can already use [fireshock](https://github.com/nefarius/FireShock)

Fivezero05, 2017-01-28:
Everytime I restarted my PC, I had to reinstall my bluetooth drivers to get it working. 
I use scptoolkit on two laptops, on with a terrible battery, so I always used the power adapter. 

I had no issues when restarten while on the power adapter, but it all stopped working when booting on battery. 

I have found a solution that fixes my problem :

For people who have an integrated Bluetooth card :
Go to "change advanced power settings"
under the tab : wireless adapter settings , change everything to max performance (especially the battery setting)

beetree, 2017-02-12:
This issue is coming up as top search result when searching for `bluetooth driver error after restart`. @Fivezero05 's answer solves a set of issues that are broader than the issues discussed in this thread.

If anyone has issues with bluetooth not working after reboot, do the following (ty @Fivezero05 ):
```
1. Open control panel
2. Search for "power"
3. Click "Edit power plan"
4. Click "Change advanced power settings"
5. Expand "Wireless Adapter Settings"
6. Choose "Maximum Performance"
```

Hope someone finds this helpful!

charlesmilette, 2017-02-12:
I would try this but now I have a new Bluetooth chip that just doesn't
works with ScpToolkit XD

mirh, 2017-02-12:
Pretty sure with zadig you can get basically just everything to work. 
I'm not sure if it was still available in last version of SCP, or you needed a slightly older one though.

charlesmilette, 2017-02-12:
The driver actually installs correctly, the Bluetooth feature just doesn't works #497

roh420, 2017-03-13:
[SOLVED]
If you go to SCP Settings Toolkit Manager you can go to service settings. Uncheck Force Bluetooth driver installation and the problem will be fixed. Try it out. :)

racompave, 2017-12-06:
Anyone who is still having this issue, just follow these steps...

1-Uninstall drivers, uninstall program, clean junk files from the installation folder ( optional: run CCleaner to delete registry keys)

2- Install this version : https://sourceforge.net/projects/scptoolkit.mirror/

This worked for me after spending 2 hours configuring the damn thing. Hopefully, it will work for you too.

tosho, 2017-12-06:
Is there a new release? I do this almost every time after I restart my PC
still doesn't work

closed 2018-01-26 by nefarius