nefarius/ScpToolkit #54: Memory leak after several hours

runderscore, 2015-11-26:
I've found that, after several hours of (in)activity, the SCPService suddenly begins consuming vast amounts of memory and CPU power.
[image]
I can't narrow down when this began to occur, but while I only noticed it originating from SCPService recently, it has been happening for quite some time.

System details follow.
>Operating System
	Windows 10 Pro 64-bit
CPU
	Intel Core i5 2500K
RAM
	16.0GB 
Graphics
	4095MB NVIDIA GeForce GTX 970
Audio
	Creative X-Fi Audio Processor 

Using a DS2, a DS3 (official), and a USB bluetooth device (BCM20702A0).

nefarius, 2015-11-26:
Holy sh*t! I will investigate, thanks for reporting!

nefarius, 2015-11-29:
I fixed it in the latest pre-release, stable one coming up next, thanks again for reporting!

closed 2015-11-29 by nefarius

Sir-Joel, 2016-04-10:
I know this thread is closed, but I'm experiencing the same issue with the most recent release.
[image]

charlesmilette, 2016-04-10:
What is the version given in the control panel?

Sir-Joel, 2016-04-11:
It says version 1.6.238.16010. I'm going to give that beta that was released today a shot though. I'll see if it exhibits the same issue.