nefarius/ScpToolkit #56: Input delay on DS4 wired+wireless

timelf123, 2015-11-29:
There is a noticeable input delay when using my DS4 controllers and a clean install of SCP Server. This does not seem to happen with my wired X360 controller

nefarius, 2015-12-10:
What version are you using? Is the delay measurable? I myself own and use a DS4 and I can't really tell a difference so I'd need more information.

timelf123, 2016-01-05:
I'm not sure how you would go about measuring it precisely, but I'd estimate ~200-300ms at the worst, which translates to over/under-compensating with the joystick

nefarius, 2016-01-05:
Ok that's awful; input updates occur *at least* every 10 milliseconds. I'm afraid I can't really help here since I'm not really convinced it's a software issue.

nefarius, 2016-04-02:
This has long been fixed; there was a problem with the initialization of the DS4 setting the input update rate too low. Now it's set to a 1000 Hz by default.

closed 2016-04-02 by nefarius