nefarius/ScpToolkit #57: ScpServer.exe spawns many threads and has high CPU usage.

ActuallyDidIt, 2015-11-30:
Running the current beta on Windows 7 shows 528 threads and constantly uses about 5-7% CPU. Is this expected behavior?

TheHardew, 2015-11-30:
On optimized beta I have 10-15% cpu and I think it's stuck in some thread-adding loop - it adds 2 threads every seconds, now I have 659 of 'em. If I disconnect my ds3 it stops and cpu usage is 0%.

OptimisedFun, 2015-12-01:
I have this too; i7 6700K. Hits ~80% cpu, thousands of threads within a second and stays there. It *works* fine but I'm sure it shouldn't be doing that.

Cauterite, 2015-12-09:
The situation seems to have greatly improved now; since I updated to v1.6.186.15342-BETA, the service is eating only a few percent of CPU cycles and sitting at around 20 threads.

nefarius, 2015-12-09:
Good to know, thanks for your feedback!

nefarius, 2015-12-12:
[Got fixed in this release.](../releases/tag/v1.6.186.15342-BETA)

closed 2015-12-12 by nefarius