nefarius/ScpToolkit #59: Controls get stuck with latest stable and beta

Keefaetana, 2015-11-30:
The latest stable and beta builds seem to have a big regression in them. When playing a game, the controls are responsive at first, but after about half a minute the controls get "stuck". A few seconds after that the controller starts vibrating for several seconds straight. When it stops vibrating, the controls are responsive for about 10-15 seconds again. Then they get stuck again and after a few seconds the controller starts vibrating again. That repeats over and over again. I tried uninstalling the latest versions and downgraded to the last stable before the current one and the controller works perfectly.

I'm using Windows 10 x64 with all the latest updates and an official DS3 that came with my PS3. GitHub doesn't seem to allow attaching .zip files, so I uploaded my logs here: http://koti.mbnet.fi/nurmip/ScpToolkit_Log-Package_Keefa_1448905405.zip

TheHardew, 2015-11-30:
Also it's only when wired and disabling rumble doesn't fix it.

DSelitskiy, 2015-12-01:
Yep, can confirm. Going back one release helped.

nefarius, 2015-12-01:
The service log is missing in your archive; did you remove it or was it never there?

Keefaetana, 2015-12-01:
I have not removed anything from the .zip file. I just uploaded it as is and didn't think something could be missing. I'll install the beta again and get the service log for you.

MonthOLDpickle, 2015-12-03:
@DSelitskiy How do I downgrade?

DSelitskiy, 2015-12-04:
Do full uninstall. Then grab the previous release from the releases page
and install that instead.

Kind regards,
Dmitry.

---
*Dmitry Selitskiy*

*email*: dmitry@selitskiy.com
*mobile*: +64 (21) 059 8239   ||   *skype*:  dmitry@selitskiy.com
*web*: http://www.selitskiy.com || http://www.thought-wired.com

MonthOLDpickle, 2015-12-04:
@DSelitskiy I am very new to using github and SCPtoolkit. I don't even see a release folder in https://github.com/nefarius/ScpServer

DSelitskiy, 2015-12-04:
Here: https://github.com/nefarius/ScpServer/releases and download this
https://github.com/nefarius/ScpServer/releases/tag/v1.6.126.15319

Kind regards,
Dmitry.

---
*Dmitry Selitskiy*

*email*: dmitry@selitskiy.com
*mobile*: +64 (21) 059 8239   ||   *skype*:  dmitry@selitskiy.com
*web*: http://www.selitskiy.com || http://www.thought-wired.com

skite2001, 2015-12-04:
seems to me, like issue #39
https://github.com/nefarius/ScpServer/issues/39

Keefaetana, 2015-12-04:
skite2001: Yes, I agree. There are three bugs about the same issue now.

MonthOLDpickle, 2015-12-04:
@DSelitskiy Welp I can't uninstall old one. I have no clue why. I did follow instructions etc. Service doesn't exist but old version says new one already installed aborting. The install folder is empty...

DSelitskiy, 2015-12-07:
There's a cleaner tool that comes with the driver, try running that.

Kind regards,
Dmitry.

---
*Dmitry Selitskiy*

*email*: dmitry@selitskiy.com
*mobile*: +64 (21) 059 8239   ||   *skype*:  dmitry@selitskiy.com
*web*: http://www.selitskiy.com || http://www.thought-wired.com

Keefaetana, 2015-12-09:
The latest beta seems to have fixed the problem for me. Excellent work. Closing the bug now.

closed 2015-12-09 by Keefaetana