nefarius/ScpToolkit #60: DS3 does not seem to work w/ iMac + Parallels

timothybasanov, 2015-11-30:
I'm trying to use Original DS3 controller to play South Park the Stick of Truth (Steam version) using Parallels 11 on my iMac. Bluetooth does not seem to work at all, controller is not seen. USB works intermittently.

During drivers installation (w/ DS3 plugged into USB of Guest OS and accessible via Bluetooth) there are 3 Bluetooth controllers: Virtual Bluetooth (probably this is host OS), and two controls (probably used for mouse/keyboard emulation). DS3 is properly visible as USB-connected device. After driver installation controller only works when connected via USB.
It would be all and well, but when I try to play and use "several buttons at once" (BR+BL or left stick up + right stick down) controller "hangs" for several seconds. It does not seem to access any input. Sometimes it starts to rumble (unless rumble is disabled) for several seconds. Replugging works, waiting it out works as well until next time I'll try to press several buttons.

I have not found any reasonable way to test my controller using Scp toolkit and not a game itself.
Hope you'll have some ideas on how to debug this issue.

Logs collected with Scp Debug Info:
https://gist.github.com/timothybasanov/4f5e64aad83d1f985503
Just in case it may be important: iMac Retina Late 2014, OS X 10.11.1, Parallels 11.1.0, Windows 10 N (freshly downloaded and updated this week).

CalculatorJones, 2015-11-30:
I expierience the exact same issue, using iMac Late 2014 and Parallels 10 with Windows 10. In addition, connecting 2 Controller is tricky, they can't be connected both via Bluetooth, and it needs a procedure of disconnecting and connecting them via cable in a certain order to work at least 1 BT and 1 wired.

timothybasanov, 2015-12-01:
@CalculatorJones Any chance you can share how you made bluetooth work? For me controller in not detected at all.

CalculatorJones, 2015-12-01:
I messed around with all my Settings, but the final steps which made it work were the following:

1. Install the Drivers. 

For Bluetooth, select the Bluetooth USB Host Controller, (to be able to select this one, after clicking on your Parallels Windows, in the Mac menu at the top of the screen you can go for devices > external devices > tick Bluetooth USB Host Controller). Maybe selecting the Virtual Bluetooth Controller created by parallels also works.

Important: This will cause the Bluetooth Controller to switch over to Windows, so you will lose every bluetooth connection to your mac, eg. Apple Mouse and Keyboard, and it will comeback when you either untick the controller in the menu or shut down the virtual machine.

For DualShock 3 Driver, select the Playstation3 Controller while its wired.

2.  Restart your computer.

3. Reset your controller with the button on the back. Then, connect your controller via cable. Check if it is detected as Controller in the SCP Monitor (buttons reacting?)

4. Detatch the cable of the controller. It should automatically connect via bluetooth. For further instructions, i used the help on the parallels website.

One thing I have to notice - after installing the bluetooth driver of SCPToolkit, i don't have any Bluetooth menu in windows anymore, but it will still connect to your Controller.

5. To connect 2 controller. Do the following:
- wire connect Controller 1
- Detatch the cable. It should be connected with bluetooth now.
- Wire connect Controller 2. At this point, both Controller stop working.
- Detatch the cable of Controller 2 and wire connect Controller 1. Now both should start working at the same time, one with cable and one with bluetooth. If you detach the wire, both will stop working.

closed 2018-01-26 by nefarius