nefarius/ScpToolkit #61: GTA V startup crashes

sevenleo, 2015-12-01:
GTA V startup crashes if scpdriver is enabled, is necessary disable, start the game, after the game is ful load press ALT+TAB and start scpdriver.
Why?
All the games i've tested worked fine, but this (my favorite) not.

nefarius, 2016-04-02:
I now own GTA V too so I can test this myself in a few days hopefully.

The-Euphoria, 2016-05-22:
@nefarius - Is there any update on this??? As my report is also regarding GTA V. My report: https://github.com/nefarius/ScpToolkit/issues/311

The-Euphoria, 2016-05-22:
Never mind, found the cause of the issue and the fix :)

nefarius, 2016-07-16:
@The-Euphoria mind to share the solution?

The-Euphoria, 2016-07-16:
@nefarius - Disable 'Vibration' in-game, that's all it was.
I read a lot of posts of people having issue with the vibrate/rumble in some games, so I disabled vibration in the in-game menu and Bingo, no more issues, re-enable vibration and instant CTD. I got to speaking with someone afterwards, he said he'd discovered the current Scp driver has an issue with vibration in GTA and some other games causing it to freeze all input while vibrating, or completely crash the driver causing game crashes and even BSOD, which effected me multiple times.

Hope this helps

closed 2018-01-26 by nefarius