nefarius/ScpToolkit #62: 2 and more BT controllers

vaseker, 2015-12-01:
Is it possible to connect 2 or more PS3 controllers via BT connection at the same time?
Thanks.

vaseker, 2015-12-01:
In my case I have 2 paired controllers, but only one (first powered on) can be connected via bluetooth.

wipeDx, 2015-12-01:
I have this problem, too.
Does the first one also stop to work when you try to connect a second one?

donutfart, 2015-12-01:
I'm also having this problem. The first controller continues to work (wireless bluetooth) after plugging out but the second one only works while plugged in. The second controller can't be paired wirelessly.

fibrizo-raziel, 2015-12-02:
I have exactly the same problem using ASUS BT400 and DS3 Siraxis Controllers. Are we doing anything wrong with the installer?
However, the old version of the drivers works great with the same hardware. I share the link for those people who don't have any other option:
http://forums.pcsx2.net/Thread-XInput-Wrapper-for-DS3-and-Play-com-USB-Dual-DS2-Controller

vaseker, 2015-12-02:
@fibrizo-raziel I have exactly same hardware as you and tried to install PS3 driver for both contorllers but it has no result because of same PID/VID.

@wipeDx first one continues work via BT if second one connected via USB. If disconnect from USB and turn off both controllers then power on them in reverse order, second controller connects via BT and first one will infinitely search BT connection.

Sorry for my poor English, guys!

Mickemoose, 2015-12-05:
I'm using the Kinivo BTD-400 with 2 DS3 Sixaxis controllers, and I'm also unable to connect more than 1 controller via bluetooth. Connecting multiple via USB works just fine though.

Huntakillaz, 2015-12-10:
Oh good to know this is an issue, spent the last hour trying to figure out whats wrong.

namachieli, 2015-12-20:
I have this same issue and same hardware profile. 2 SIXAXIS and ASUS BT400.

nefarius, 2016-04-02:
Is this still a thing?

wipeDx, 2016-04-02:
I dont think it is, I haven't had any problems with the newest versions anymore

nefarius, 2016-04-02:
I'll allow myself to close this then.

closed 2016-04-02 by nefarius