nefarius/ScpToolkit #64: Game Recognizes Controller but when i press buttons nothing is happening.

Ichigo07, 2015-12-01:
Well if i start a game and press a button on the controller the menu icons are switching to xbox symbols after that nothing else happens its like the buttons are not mapped even tho i checked them via Controller setting on windows.

alfredox123, 2015-12-02:
check if it is working as second player maybe

Ichigo07, 2015-12-02:
tried that but didnt work, i just uninstalled the new scp + drivers + uninstalled the controller and tried to reinstall the old SCPServer but i have the exact same problem. 

[Upadte:] It seems to be game related because Fallout 4 works without any problems but on The Crew and Just Cause 3 its the problem mentioned above. Weird

closed 2016-04-10 by nefarius