nefarius/ScpToolkit #65: cant play wireless

alfredox123, 2015-12-02:
i have a fake ps3 controller, on windows shows as "PS(R) Ga`epad"

i can make it to work wired but not wireless

on scp monitor says it is reserved when is unplugged and in scpserver seems to connect but it says something about hiv connection failed or something like that

i have ScpToolkit v1.6.133.15324 and in scptoolkit settings i enable the option of counterfeit controller, stills doesnt work wireless

dont know what file should i attach here to make see the problem or something

alfredox123, 2015-12-02:
how do i look if my bluetooth dongle is compatible?

Kalet, 2015-12-08:
Hi! I have the same issue here! Whenever I try to pair the Controller (DS3) the LEDs shut off and to the SCP Server shows the controller connected, but Windows 10 doesn't not recognize any button
PD:My controller is also a fake copy

alfredox123, 2015-12-08:
i think scptoolkit doesnt support fake ps3 controllers wireless as for now :(

Kalet, 2015-12-09:
The funny thing is that with an older versión it did work, but I was using Windows 7. I couldn't make any version work in windows 10 with my controller though

alfredox123, 2015-12-09:
what version exactly did you use?

nefarius, 2015-12-09:
1.2.x is Scarlett's version, not mine. Since it works entirely different I don't support it.

Kalet, 2015-12-09:
Oh I didn't know! Anyways I can't make the controller work on the latest version here in GitHub, it looks connected to SCP Server its shows Bluetooth full but the LED in the controller shut off and windows doesn't recognize any button, only the PS Button is recognized! (It's odd because with cable it works just fine!)

alfredox123, 2015-12-09:
i have windows 7 gonna test that version and see if it works wireless, are sure is that version?

Kalet, 2015-12-10:
https://mega.nz/#!2Z9UQRiL!PKL5iS0aJj6dEvqyDhYDmkFx8fQgXY4rtVnbABRx49s I downloaded it from here!

closed 2018-01-26 by nefarius