nefarius/ScpToolkit #66: L and R trigger not working.

jimmycc2, 2015-12-02:
I have a Dualshock 3. I had tried  all scp toolkit version but my L and R trigger don't work when I am using it with Bluetooth.

Rossco1337, 2015-12-15:
Same bug. Using Bluetooth reciever USB\VID_0A12&PID_0001. All functions of DS3 work when controller is plugged in through USB but when unplugging it, the "Left trigger" input locks to a value of 0.754 and the "Right Trigger" is unresponsive. All other buttons and sticks work through Bluetooth.

Tested in PCSX2 1.3.1-1540 with Lilypad 0.11. Driver installer version 1.6.133.15324.

Zareli, 2015-12-16:
I have the exact same issue. I can't take logs with the control disconnected though. In the ScpToolkit Profile Manager it registers correctly so I suppose the xbox driver is taking another input as the Z-axis (maybe some sensor from the sixaxis).

truetablecat, 2016-01-05:
Yeah, I can verify this. I'm using the latest build, and everything works great, except the L and R trigger doesn't work when using Bluetooth.

judebassil, 2016-02-05:
I can verify this issue, same for me.  Fine over USB, unresponsive triggers over Bluetooth.

ansraliant, 2016-06-18:
Same issue, hardware id: USB\VID_0A12&PID_0001&REV_0134.

ansraliant, 2016-06-18:
Same issue, hardware id: USB\VID_0A12&PID_0001&REV_0134. Driver Version: 1.6.238. All other buttons working perfectly.

nanotechnician, 2016-06-27:
Same problem here, got to say that this works perfectly with Steam Link & Virtualhere, bluetooth does not recognize L and R triggers, USB works perfect...

closed 2018-01-26 by nefarius