nefarius/ScpToolkit #67: L2 and R2 Triggers won't work on bluetooth

Zareli, 2015-12-02:
Hi, thanks for this amazing tool, everything works correctly when plugged in. When unplugged the controller connects to bluetooth within seconds and all the buttons work except the Z axis (mapped for L2 and R2 shoulder triggers.

I ran the debug tool unplugged but no debug files came. I also noticed in the gamepad input tester the L2 and R2 simply don't register.

Hardware: 
Genuine PS3 Controller
Bluetooth Dongle VID (0A12) and PID (0001)
Windows 10

Zareli, 2015-12-07:
With the new BETA updates I tried the remapping tool.
When I'm on BT the remapping tool shows all the buttons registering correctly (including RT and LT). However the Windows Gamepad Settings still shows the Z axis stuck and not registering RT/LT. I tried remapping the buttons to other gamepad buttons. Physical RT/LT will function as other buttons but other buttons won't register when mapped to function as RT/LT. Mapping any button to keyboard/mice commands doesn't work.

New Log with the latest BETA
[ScpToolkit_Log-Package_Zareli_1449532694.zip](https://github.com/nefarius/ScpServer/files/54754/ScpToolkit_Log-Package_Zareli_1449532694.zip)

Auronsama, 2016-01-01:
Got the same issue, have no idea what causes this.

Fhorazombie, 2016-01-15:
I have the same problem,i try reinstalling the program but nothing
Got the same setup

Hardware: 
Genuine PS3 Controller
Bluetooth Dongle VID (0A12) and PID (0001)
Windows 10

[ScpToolkit_Log-Package_fhora_000_1452900728.zip](https://github.com/nefarius/ScpToolkit/files/92668/ScpToolkit_Log-Package_fhora_000_1452900728.zip)

Zareli, 2016-01-16:
I can confirm is the Bluetooth Dongle VID (0A12) and PID (0001) REV_0134. It has never worked in any version, maybe there's a hardware limitation in that revision (there's info in the old forum post), It appears it's not capable of BT 2.1 and above protocol.

I bought different bluetooth dongle and now all the buttons and triggers work.
It's also VID (0A12) and PID (0001) but the revision is REV_8891. It's supposed to be true BT 4.0.

Ludancv, 2016-02-17:
Im using a bluetooth BT-8001-BK, it says it supports V2.0 with 3Mbps, I'm having the same issue, While connected the control works fine, but wireless que triggers dont respond, any help?

damianshrinivas, 2016-05-19:
Sam issue here on dongle VID (0A12) and PID (0001). When connected via USB everything is working great but when I switch to bluetooth - triggers won't work and all other buttons are working fine.

Zareli, 2016-05-20:
I don't know why people is still commenting about this, this is already solved people.
V2.0 is insufficient for 6-axis wireless (it requires at least V2.1). There's a more than one dongle that match the VID 0A12 and PID 0001, we already know the ones that are not compatible. Please read before posting nonsense or at least report the REV number of your dongle if it doesn't work.

TL;DR If your triggers don't work, your dongle is not compatible, get a different one.

devondale16, 2016-07-06:
Hello Everytime I try to open profile manager, I get fatal error occured. Everything else works fine, i just would like to remap the buttons. Running win 10, with a ps3 controller btw. Any help would much be appreciated.

Kuroshiroh, 2016-12-17:
I can't use it too L2 and R2 buttons. What should we do? D;

csgokoko, 2018-01-15:
i have same problem with xbox one s controller 
i have installed the drivers for usb everything and buttons working fine but whenever i connect via Bluetooth RT/LT not working and the keys are random binds like 
A=X 
X=Y
B=A
so what should i do without upgrading my windows to 10 and use it via bluetooth
Hardware:
Genuine PS3 Controller
Bluetooth CSR 4.0 Dongle ( plug and play )
Windows 7 64 bit

closed 2018-01-26 by nefarius