nefarius/ScpToolkit #70: Rumble malfunction without bluetooth support

phly95, 2015-12-04:
I have tried installing my DS3 on a Studio XPS desktop that does not have bluetooth support and it installs fine and works fine for the most part, except rumble does not work properly. If a rumble signal is sent, the controller deactivates (as in buttons don't work) for about 10 seconds and then it vibrates much longer that it's supposed to (testing using metal gear solid settings through mednafen in retroarch and also did basic testing in dolphin settings.)

phly95, 2015-12-04:
It also happens on my laptop with bluetooth disabled.

alfredox123, 2015-12-04:
i suggest you that completly disable rumble on scptoolkit settings manager as for now, and try again on the next update

Moonshardai, 2015-12-04:
Not sure if this helps.

I had the same issue, however there was also an issue where the SCPtoolkit settings would not change settings. I could disable, even change the LEDs on the Settings page yet it would not change with the controller.

Downgraded to the last version of 1.5. the Settings worked fine and the rumble worked fine too.

wunderbrot, 2015-12-12:
I've encountered this bug as well.  Disabling rumble in scptoolkit stops the rumble function from going haywire, but the controller still deactivates whenever there would normally be a rumble.  

I have to disable rumble in-game for it to be usable.

closed 2018-01-26 by nefarius