nefarius/ScpToolkit #74: Not possible to turn off navigation controller

igorrafael, 2015-12-06:
#### Device
Original Navigation Controller (CECH-ZCS1U); a.k.a: move controller with analog stick

#### Problem
After pairing the device via bluetooth it is not possible to turn it off because the device does not have a R1 button for the PS+L1+R1 shortcut to be issued.

#### Suggestion
Custom shortcut for the navigation controller or simply an extra shortcut such as PS+X+O.

#### How I can help
I'm willing to implement this change myself but will need some help in finding the right place to implement the changes. I am a computer science major with C++ and C# experience.

igorrafael, 2017-01-27:
I never even started.

billyisms, 2017-01-31:
I haven't started using this software yet, but I hold down the playstation button for 12 seconds on the navigation controller connected to my computer through bluetooth and the controller powers down.

closed 2018-01-26 by nefarius