nefarius/ScpToolkit #76: Controller recognised but buttons not registering input.

NexxusDrako, 2015-12-07:
I have an official DS3 and while it used to work it randomly stopped working after i plugged in my MadCatz Xbox 360 wheel, and I haven't been able to get it working since. The lights just endlessly flash and the buttons don't work. I've tried everything short of uninstalling the wheel, but I'll see if it helps.

NexxusDrako, 2015-12-08:
I couldn't uninstall the wheel btw.

nefarius, 2015-12-08:
[Please try this release](../releases/tag/v1.6.186.15342-BETA) and tell me if it's fixed for you, thanks!

NexxusDrako, 2015-12-09:
It's fixed the initial issue in #77, but unfortunately buttons still fail to work. I'm going to try the fix in that post now.

NexxusDrako, 2015-12-09:
It did not fix. I posted in #77. Feel free to mark this as a duplicate.

nefarius, 2015-12-09:
Closed due to hardware problem.

closed 2015-12-09 by nefarius