nefarius/ScpToolkit #77: Problem Installing DS3 Driver

ace3df, 2015-12-07:
Tried a bunch of things (Force install, booting with USB creds off, etc.)
Kinda at a loss...
Any idea?

Log:
http://puu.sh/lN2wj/305e1aa04c.ZIP

NexxusDrako, 2015-12-08:
My ticket #76 is doing the exact same thing!

NexxusDrako, 2015-12-08:
Possibly related, when installing through the links on desktop, it creates a Drivers folder on the desktop, which I don't think belongs there.

nefarius, 2015-12-08:
[Please try this release](../releases/tag/v1.6.186.15342-BETA) and tell me if it's fixed for you, thanks!

ace3df, 2015-12-08:
It installed this time! But nothing seems to be able to find it now. 

ppsspp, psx2, dolphin, etc.

I had this problem (didn't show in the Log section):
http://puu.sh/lOhJL/7363c475b8.png

I then tried the SCP Server legacy to just try and got:
http://puu.sh/lOhOL/c4f93022b4.png

nefarius, 2015-12-08:
There's a tool called `ScpToolkit Debug Info Collector` you can find via start menu, please run it, follow it's instructions and forward me the result.

ace3df, 2015-12-08:
I uninstalled the USB hub through device manager (Clean Wipe Utility didn't seem to go for reinstalling the PS3 controller like before) - and now it finds the Playstation 3 controller (Dolphin) but no buttons seem to work.

Log:
http://puu.sh/lOiPo/e43b246fc5.zip

nefarius, 2015-12-08:
Don't run the debug collector after clean wipe, you'll distort the results. Make a clean installation, connect the controller(s), shut down the service via `services.msc` and run the debug collector again.

nefarius, 2015-12-08:
Could you also please try [this special build](http://www.xup.in/dl,16542130/ScpControl_1.6.188.15342.ZIP/), just overwrite the existing DLL and restart the service. Submit the logs afterwards.

NexxusDrako, 2015-12-09:
https://dl.dropboxusercontent.com/u/62181279/ScpToolkit_Log-Package_James_1449667611.zip

nefarius, 2015-12-09:
Could you explain in more detail what' exactly happening on your end? All logs look perfectly fine so I'm a little lost what to recommend now...

NexxusDrako, 2015-12-09:
Well, I am plugging in the controller,and when the service is on, it will recognise it. However, the lights on the controller are flashing as if the controller is still syncing. When I go into the controller setup screen where it has the buttons and axis input checker, nothing is working. I press a button and nothing will light up. I move an analog stick and the axes don't move. Meanwhile, my x360 wheel and x360 controller work fine.

nefarius, 2015-12-09:
Could you provide a screenshot of your "Devices & Printers" section along with your device manager, expanding the USB and HID categories?

NexxusDrako, 2015-12-09:
https://gyazo.com/f9c5d2417eb6968763b4869713d69a7f < Devices and Printers

https://gyazo.com/4bf35eddd7bb2f0174a2c395367a6f63 < Device Manager

nefarius, 2015-12-09:
Ok, screenshot of `ScpMonitor` please.

NexxusDrako, 2015-12-09:
https://gyazo.com/e7bc3406232ed20afc3f5e697b537c4c

nefarius, 2015-12-09:
Thanks, getting more and more mysterious... @_@

NexxusDrako, 2015-12-09:
I just plugged in my brother's PS3 controller and it worked fine. I'm not sure what's going on anymore.

nefarius, 2015-12-09:
Give me a few secs, have and idea...

NexxusDrako, 2015-12-09:
If it involves using the reset button, I just thought of it. heh

nefarius, 2015-12-09:
Nah, code-based idea =D

NexxusDrako, 2015-12-09:
Haha, excellent.

nefarius, 2015-12-09:
Please test again with [this one](http://www.xup.in/dl,15293666/ScpControl_v1.6.189.15343.7z/), should pop up an additional error message in your case (hopefully).

NexxusDrako, 2015-12-09:
https://gyazo.com/190920034256699c22f6c775f7855283

nefarius, 2015-12-09:
Ok that's not good. This is a very low level error I can not fix by coding. Can you maybe try a different USB cable? Is the USB port on the controller somehow wiggly or visibly damaged?

NexxusDrako, 2015-12-09:
The port is very wiggly and indeed damaged. Looks like I've gotta get a new one. heh

I didn't even notice it before. :(

nefarius, 2015-12-09:
Are you skilled with a soldering iron? Re-flowing solder can be magic :)

NexxusDrako, 2015-12-09:
Unfortunately not. I can program, but that's about it. haha
Thanks for your help, anyway.

nefarius, 2015-12-09:
You're welcome :)

nefarius, 2015-12-12:
Any news from ace3df since we hijacked your thread? :)

closed 2016-04-02 by nefarius