nefarius/ScpToolkit #78: Trying to get games besides Super Meat Boy to detect my Afterglow PS3 controller.

Narkah, 2015-12-08:
Copied from my forum post.

So, first of all, just wanted to thank you for updating this wonderful tool! I made my account just to post here. So, I've been having quite a few issues with my Afterglow PS3 controller. AFTERGLOW AP.2 Model: PL-6322. Don't know if you need the PCC ID. Anyway, I've been able to get it to run on one game, the only game I really needed it running on, which was Super Meat Boy. The sticks work, and the shape buttons (EX: X,O | Proper terminology, please?). On other games that actually detect the controller, the buttons are sometimes "rotated". EX: X is O, O is triangle, Triangle is square, square is X. I also recently bought Skyrim for PC, and I'm trying to get it to work with my controller. Skyrim doesn't even detect it. On other games, like Mercenary Kings, the buttons are "rotated", and in Outland, the triggers are acting like Select and Start.

Thanks in advance!

-Narkah

Narkah, 2015-12-10:
Here's the debug thing. Missed that teh first time I read the post. :P

[ScpToolkit_Log-Package_Narkah_1449740714.zip](https://github.com/nefarius/ScpServer/files/57915/ScpToolkit_Log-Package_Ross.Filbin_1449740714.zip)

nefarius, 2015-12-10:
Well, since you didn't install the Afterglow driver according to the log it's no surprise it doesn't work... ;)

Narkah, 2015-12-10:
"Afterglow Driver according to the log"... I didn't see anything about an afterglow driver... how do I install that? I'll go digging through all of the documentation I can to see what I missed. :P

nefarius, 2015-12-10:
The reason it worked with some games and others not is, that there game pads get detected by games using DirectInput.

Narkah, 2015-12-14:
Well. The controller appears to have stopped working entirely, I must have botched the install somehow. I tested it on games that were previously working. They aren't anymore. Hmmph. Here'
[ScpToolkit_Log-Package_Narkah_1450082972.zip](https://github.com/nefarius/ScpServer/files/60958/ScpToolkit_Log-Package_Narkah_1450082972.zip)
s the new debug.

closed 2018-01-26 by nefarius