nefarius/ScpToolkit #79: [LilyPad SCP] Can't config all stick axes. Result in calibration problem.

skite2001, 2015-12-09:
I cant config the "down" and "right" axes of the L and R -Stick. It happens nothing, if i try to set these to the fitting gui-buttons. can't play this way, because my character turns in circles :(
[image]
[image]

nefarius, 2015-12-10:
Right, you've spotted a bug. I'll work on a fix!

nefarius, 2015-12-12:
I fixed the issue; playing Metal Gear Solid 3 right now :) Will be included in the next release!

skite2001, 2015-12-13:
very much thx. and good work

closed 2015-12-13 by skite2001