nefarius/ScpToolkit #80: Controls blocked when playing a game

Kelarang, 2015-12-10:
I was trying to play Rocket League after the installation of this new program. All fine, i see the n°1 led blinking thinking it was all fine. I launched the game, used my directional arrows to choose what i was gonna do, then i presser R2, and Circle to start playing then the controller froze, and my car was keep going forward. What can i do? I've not a single idea on how i can fix this

nefarius, 2015-12-10:
The Log Pack would be even better ;)

Kelarang, 2015-12-10:
Where do i check it?

nefarius, 2015-12-10:
Search for `ScpToolkit Debug Info Collector` in your start menu, run it and follow its instructions.

Kelarang, 2015-12-10:
Ok, found it! There ya go! Hope it's fixable, Xbox's controllers are just so uncomfortable :/

Kelarang, 2015-12-10:
What kind of file do you need? The all the pack?

nefarius, 2015-12-10:
The created ZIP.

Kelarang, 2015-12-10:
Got winrar which is not supported, gimme a sec :P

nefarius, 2015-12-10:
Just use any of the 2 million one-click upload sites :D

Kelarang, 2015-12-10:
Got no idea on how to do it :/ I'll try to create a zipped one as soon as i come back. Im sorry, but i really need this to be fixed :/

nefarius, 2015-12-10:
The tool creates a ZIP file. Take this file and upload it to e.g. http://xup.in and paste the link here.

copain9, 2015-12-10:
Hi. I'm having the exact same problem and also in Rocket League. I was using an older version of SCP server and at some point I had issues with some input not registered or some bad input registered (I had dpad registerd when I pressed too much button at the same time). So I installed the last version (very nice one btw) and I started having the same problem as OP. 

Buttons get stuck when pressing after a while; they get released ~20 seconds later. and this process repeats just after.

My debug file : http://www.xup.in/dl,96566262/ScpToolkit_Log-Package_Greg_1449777297.zip/

copain9, 2015-12-10:
The problem also happen with Dishonored. I did some experiments. The stuckness/rumble is a sequence. I timed 9-10 seconds between the moment the controls get stuck and the moment the rumbles activate (controls still stuck). 10seconds after the rumbles stop and the controls unstuck. After that, the time before is gets stuck again varies, it can get from only 1 input to a few ones.

Kelarang, 2015-12-10:
Op is back:

This is my Log Pack http://www.xup.in/dl,16853221/ScpToolkit_Log-Package_Dani_1449767170.zip/
copain9 added a quite accurate description of the problem. 
Your name isn't new tho, i think i've already played with ya

nekogami, 2015-12-10:
I think it is may be linked to #39

copain9, 2015-12-11:
As mentionned in the other report, installing ScpToolkit v1.6.186.15342-BETA seems to fix my issue (tried 2 minutes).

Kelarang, 2015-12-11:
Hm where can i download it?

nefarius, 2015-12-11:
[Seriously?](../releases) =P

Kelarang, 2015-12-11:
Sorry for my ignorance, im not used to this things :/

Kelarang, 2015-12-11:
(Ty for the link tho)

nefarius, 2015-12-11:
Np ;)

nekogami, 2015-12-11:
Got some problem yesterday with the latest beta (cf: last post in #39 for description and available fallback)

Kelarang, 2015-12-11:
Haven't had the time to thank you for continuing this project, creating an easy way to let us play with our controllers. Thank you!

closed 2018-01-26 by nefarius