nefarius/ScpToolkit #81: Cant pair DS4 with bluetooth dongle

h00, 2015-12-11:
I'm just re-entering the windows world, so I may be overlooking something obvious here. The bluetooth dongle I am using is also not listed as one of the verified USB devices (Device USB\VID_19FF&PID_0239\98588A0514BE). I installed ScpToolKit, the BT dongle drivers and the PS4 drivers. I then disabled the service and ran the stand alone server which dumped these logs http://pastebin.com/NxkWLLtq. Let me know if you need more info.

nefarius, 2015-12-11:
Search for `ScpToolkit Debug Info Collector` in your start menu, run it and follow its instructions.

h00, 2015-12-11:
Thanks. For some reason github isn't letting me attach the zip here, so I have uploaded it to https://mega.nz/#!Lp0R1YhR!PBWiqAXkPC-hnElO1jqf1Oa5f1Jjxf2pBZrFWErCbDQ.

Xelbair, 2015-12-20:
[ScpToolkit_Log-Package_Xelbair_1450611928.zip](https://github.com/nefarius/ScpServer/files/67790/ScpToolkit_Log-Package_Xelbair_1450611928.zip)

I have exactly same problem - except i'm using supported bluetooth dongle.

IlianBurgos, 2016-02-09:
Hi, h00, did u find a solution? I got the same problem with my ds4.

masterx193, 2017-01-05:
If you paired normally before, you have to uninstall it before 
USBDeview helps nicely http://www.nirsoft.net/utils/usb_devices_view.html

closed 2018-01-26 by nefarius