nefarius/ScpToolkit #82: Dualshock API extensions

mirhl, 2015-12-11:
After having read [this](http://steamcommunity.com/groups/frameratepolice/discussions/0/535151589883097165/#c487876568226598506) I wondered if this driver (already a must for DS3 if you ask me) couldn't do something to improve the situation. In particular for unorthodox features like the light bar and touchpads. 

Contrarily to #6 though, there are the default DS4 hidden (yet exposed) implementations that shouldn't just break imo. 

So I summon you, hoping not to piss somebody: @Marioysikax, @flibitijibibo, @Jays2Kings

nefarius, 2015-12-11:
Huh? I read the links but I - excuse my ignorance - didn't quite get what you want from me? :)

mirhl, 2015-12-11:
:P
At the moment light bar and touchpad access in Windows are hacks at most. 
A more standardized way (aka APIs, like those for sixaxis and pressure sensitivity) would be great.

nefarius, 2015-12-11:
Well, wouldn't be too hard for me to provide one. Just needed to know what developers would need.

flibitijibibo, 2015-12-12:
Right, sorry I didn't get to this sooner...

This is basically in reference to FNA's SetLightBarEXT extension, which currently looks like this:

https://github.com/flibitijibibo/FNA/blob/master/src/SDL2/Input/SDL2_GamePad.cs#L620

Basically just a path only used by Linux runtimes when we have write access to the files in /sys/class/leds, but we've not had a Windows path available to us. It sounds like there are additional libs we can depend on to do this, and we could probably check for a C entry point at load time on Windows to determine if we have access to the light bar or not.

I'd be totally okay with just a `SetLightBar(u8 r, u8 g, u8 b)`, the only real concern is how to match up light bars with DirectInput devices (or more specifically, SDL_GameController devices that are using DirectInput underneath).

nefarius, 2016-08-29:
@mirh I designed FireShock with those requirements in mind; it will become mind-blowing(ly) easy to access those features through the driver :wink:

closed 2018-01-26 by nefarius