nefarius/ScpToolkit #84: ScpServer installs on wrong USB interface number

fdavison, 2015-12-12:
ScpServer is installing on wrong USB Interface on QuadStick.

The QuadStick has four USB interfaces:
   Interface 0:  GamePad
   Interface 1:  Mouse
   Interface 2:  Keyboard
   Interface 3:  Flash Drive

All four interfaces are showing up in the Dual Shock 3 controllers list in the DriverInstaller.  Selecting interface 0 and running the installer results in a different interface showing up under libusbK in the Device Manager.   (In this latest run, interface 2) The interface number is not always the same with each test. In joy.cpl, the QuadStick still shows up and is functional, the XBox 360 controller is not.

The wrong interface shows up in the Ds3Controller_.....inf file:

DeviceID   = "VID_16D0&PID_092B&MI_02"

so the mistake seems to be happening before that file is generated. 

Logs, etc: https://www.googledrive.com/host/0B7MVckMP_1T0U2JNWG5aTE5odkU

Jan200101, 2015-12-12:
I think that might be happening because of the naming. Interface 0 is named Quad Stick in the Installer and in the Device manager Interface 2 is too called Quad Sticks.

nefarius, 2015-12-12:
Might have identified the problem, please [test and provide feedback on this release](../releases/tag/v1.6.196.15346-BETA).

fdavison, 2015-12-12:
Now the correct interface 0 is displayed, and the QuadStick properly disappears from the Joy.cpl list, but there is an exception being thrown when starting the service.  If I unplug the quadstick, I can start the service, but when I plug the quadstick back in, an error message pops up. 

Logs: https://www.googledrive.com/host/0B7MVckMP_1T0V2E3eU9VM2trcE0/

fdavison, 2016-01-14:
It appears to be installing on the correct Interface number.  I will close this issue, however I'm still having trouble with the service throwing an exception and stopping whenever the quadstick is plugged in.

closed 2016-01-14 by fdavison