nefarius/ScpToolkit #85: Constant error

DracoTempus, 2015-12-13:
Don't know what changed. I was on the 1.5~ version for a while and all was working till earlier.  I was at work so I don't think anything changed.

Updated to the 1.6 (looks nice btw).  Still same error.

I don't know what the logs show so some info here:
Windows 10
AC-7260 bluetooth adapter

I tried the ScpCleanWipe and it didn't work for my issue.  I uninstalled all the drivers and reinstalled.  I tried not installing the service and just using the server.  I am able to plug my controller in and play....but still no bluetooth.

[ScpToolkit_Log-Package_FamilyComputer_1449977324.zip](https://github.com/nefarius/ScpServer/files/60326/ScpToolkit_Log-Package_FamilyComputer_1449977324.zip)

[image]

Jan200101, 2015-12-13:
same but for me it wont work at all without starting the SCPServer

DracoTempus, 2015-12-13:
Yep, that is the SCPServer.  If I plug in a controller that controller will work.  However, as you can see the bluetooth isn't connecting for some reason.

Jan200101, 2015-12-15:
No I mean my controller will not connect at all without starting SCPServer its really strange

nefarius, 2015-12-15:
There's no service log. Installer log also states you didn't install the service. The server is deprecated, install the service and try again.

DracoTempus, 2015-12-15:
I thought you are not suppose to have the service running if you are running the SCPServer.exe. 

If the service is running it won't open SCPServer.exe.

However,
Yeah sorry I tried installing the new server prior and same issue.  I changed the bluetooth adapter and it still had the same error.  

Although, I did get 1.5.x.x working with the other bluetooth adapter.

nefarius, 2015-12-15:
Yes that's correct **but the server is only for testing purposes**, the service is the way to go. And as the error message states they can and must not run at the same time.

DracoTempus, 2015-12-15:
Ahhh I see. Ok I will try the service and report back.

Jan200101, 2015-12-17:
For some reason i cant start the Service after a restart

closed 2018-01-26 by nefarius