nefarius/ScpToolkit #86: Can't get USB driver to install on ANY computers, running Windows 10

Ganz-hun, 2015-12-13:
Hello!
I tried to install the BT software on my Windows 10 laptops (Asus K455L, and a first gen. Surfce Pro 128GB), using the bult-in Bluetooth device, and it says WDI_ERROR_OTHER. I also tried a 3rd party USB BT dongle, and it installed fine on my Windows 7 PC, but not on the computers running Windows 10. Unfortunatelly, I couldn't find the debug app, so I attached two screenshoots. I hope it's enough.

[image]
[image]

Ganz-hun, 2015-12-14:
The new version installed fine :)

nefarius, 2015-12-15:
Good to know, I'll allow my self to close this issue then :)

closed 2015-12-15 by nefarius