nefarius/ScpToolkit #87: Issue with connecting multiple DS3 controllers using Bluetooth

KraazyKen, 2015-12-13:
Hello,

Your drivers works very well but I'm only having an issue when I connect multiple DS3 controllers using Bluetooth. To start I have selected both of them when I installed the DS3 drivers and I also installed the Bluetooth drivers as well. When I have two DS3 controllers connected via USB they both are detected and works well. I then remove one from the USB and they both still work, one via USB and one via Bluetooth. Now here is where the problem is. When I remove the second USB and try to pair them both up via Bluetooth, they BOTH stop working completely. Neither controllers responds to any buttons anymore and for me to fix it I would need to connect them via USB again. I would want to be able to use more than one controller with Bluetooth but because of this issue I can only use one while the rest has to be connected using USB. 

Is there any way to fix this? I actually did NOT have this issue before but it only happened when I downloaded the newer versions. (It was about a month ago) I downloaded the newest BETA (ScpToolkit v1.6.186.15342-BETA) hoping it would fix this issue but the problem still persists.
I have attached debug log is this will help. Please let me know if you can fix this problem I'm having.
Thank you.

I am using Windows 10
I am also trying with two Official Sony Dualshock 3 Controllers
My Bluetooth Dongle is the Kinivo BTD-400 Bluetooth 4.0 USB Adapter

[ScpToolkit_Log-Package.zip](https://github.com/nefarius/ScpServer/files/60398/ScpToolkit_Log-Package.zip)

espinozagabe, 2015-12-14:
I am having the exact same problem. Any help is appreciated.

namachieli, 2015-12-15:
Same problem for me. Same with newest and previous version.

Angelorpheus, 2015-12-18:
same problem too :(

Gktman, 2015-12-22:
I have the exact same problem with 2 different dongles

seminarian, 2015-12-27:
Thanks Gktman. I'm using the old version u shared flawlessly with my belkin USB dongle. I also had the exact same problem with this dongle. (only 1 controller possible on bluetooth)
I had to use zadig to install the bluetooth driver and I needed to add scpserver.exe to my windows startup folder because scpservice doesn't install, but now I can play every time!

For my integrated intel bluetooth it also works with scpserver.exe but when the program is closed and restarted it gives "host error". This can be fixed by running zadig again. (this is necessary everytime the program was shut down)

So I'm happy I'm just using my belkin usb dongle as dedicated bluetooth and my integrated laptop to connect with my mobile phone.

Thanks alot for the share! You relieved me from much pain ;-P

sceithamer, 2015-12-29:
I am also having the same problem.  I am only able to connect one controller to my bluetooth dongle.  I will try the old version and see if that fixes my issue as well.

closed 2016-01-04 by KraazyKen

espinozagabe, 2016-01-04:
Why did you close the issue? I'm still having this problem.

KraazyKen, 2016-01-04:
I edited my first post but I'll make a new post about it too.

As of version v1.6.229.15365 this issue has been resolved! Multiple controllers are now recognizable through Bluetooth. Thank you for fixing this Nefarius, I will now close this issue.

espinozagabe, 2016-01-04:
Oh, awesome! I didn't see a response from Nefarius here so I just assumed it wasn't fixed. I'll update to the newest version tomorrow and report back.