nefarius/ScpToolkit #89: Button prompts

Tomgroot, 2015-12-15:
When I connect my PS4 with USB via the ScpServer the xbox button prompt shows up in all games.
[image]
[image]

If I connect my PS4 controller via bluetooth, my controller does not show up in the ScpServer. Obviously because I didn't install the drivers for the bluetooth connection, but when I do so my games HAVE the PS4 prompt! Is there a way to let the games think you are using a actual PS4 controller instead of a Xbox controller, in other words is it possible to show the PS4 prompts in games that support these?

[image]
[image]

Tomgroot, 2015-12-15:
Correct me if I am wrong, but I think I am using Dinput when I use the bluetooth without Scp and Xinput when I use the USB with Scp. So is there a way to support Dinput in the future?

nefarius, 2015-12-15:
Since the driver emulates an Xbox Controller: no, it's not possible to change the button overlays. And no; the game doesn't know about Bluetooth or USB; it sees an Xbox controller and therefore uses XInput, not DirectInput. [There is a 3rd party mod for DInput though I haven't tested it myself](http://forums.pcsx2.net/Thread-Dinput-Wrapper-for-SCP-Driver).

closed 2015-12-15 by nefarius

kq01526, 2016-01-09:
> @nefarius wrote:
> 
> it sees an Xbox controller and therefore uses XInput, not DirectInput

That is plain wrong.

When you connect a DualShock 4 to a Windows PC via USB or pair it via bluetooth without using any additional driver or wrapper, then the DualShock 4 is being recognized as a DirectInput gamepad.

There is an increasing number of games which support the DualShock 4 out of the box (no additional driver or wrapper needed) as a DirectInput device and those games recognize that it is a DualShock 4 and they show PlayStation button prompts instead of Xbox button prompts. Assasin's Creed: Unity is among those games.

If you want to read more about it, see the following thread for example:

http://www.neogaf.com/forum/showthread.php?t=935140

So, @Tomgroot is perfectly right in saying that he was playing in DirectInput mode when Assasin's Creed: Unity (and the other game) was showing PlayStation button prompts instead of Xbox button prompts.

What he was asking for was if you could add a DirectInput mode to ScpToolkit, which would allow to switch from XInput to DirectInput, to be able to play those game that support it to run with DirectInput and to show PlayStation button prompts. Because using ScpToolkit of course defeats the out of the box DualShock 4 support of those games because using ScpToolkit they see a XInput device instead of a DirectInput device and show Xbox button prompts instead of PlayStation button prompts.

@Tomgroot :

What is the other game on the second screenshot showing the car?

nefarius, 2016-01-09:
@kq01526 what the hell are you talking about? When you use a DS4 through Scp via `libusbK` drivers the DS4 as a device becomes invisible to any apps/games that either use DInput, RawInput or XInput. It's Scps virtual Xbox 360 controller driver that gets virtually plugged in and is seen by the game through xusb drivers (Microsoft) and XInput-API. So the game "sees" that an Xbox controller is used; it has no knowledge what it really is.

Yes, I know how the DS4 works in its "native-mode". Yes, I know that some games support the DS4 directly. What has any of this to do with the questions @Tomgroot asked? I think I already answered them properly...

kq01526, 2016-01-09:
> @nefarius wrote:
> 
> What has any of this to do with the questions @Tomgroot asked? I think I already answered them properly...

He told/asked you this:

> @Tomgroot wrote:
> 
> Correct me if I am wrong, but I think I am using Dinput when I use the bluetooth without Scp and Xinput when I use the USB with Scp.

You replied:

> @nefarius wrote:
> 
> it sees an Xbox controller and therefore uses XInput, not DirectInput.

Which is wrong isn't it? If the game would see an XInput device (Xbox controller), then it would not show PlayStation button prompts. But the screenshots clearly show the game using PlayStation button prompts, so it sees a DirectInput device, not a XInput device, so you are wrong.

nefarius, 2016-01-09:
My answer is related to him using the Scp drivers. Why would I wanna comment on the DS4s native mode?

kq01526, 2016-01-09:
> @nefarius wrote:
> 
> My answer is related to him using the Scp drivers. Why would I wanna comment on the DS4s native mode?

Because he (and me) is asking you for a way to use the native mode without having to uninstall ScpToolkit.

nefarius, 2016-01-09:
@kq01526 since this would defeat the entire purpose of the software as I'd have to emulate a DS4 just for those DS4-aware games; no, there is no way currently and I doubt I'll follow this trail since it's not what the software is meant for. Good enough?

kq01526, 2016-01-09:
> @nefarius wrote:
> 
> just for those DS4-aware games

It still would be woth it, since having PlayStation buttons is a big plus.

Well, besides, [DS4Windows](http://ds4windows.com/) also has integrated an option to switch to DirectInput, see the "Use DInput only" checkbox in the lower right corner of the following screenshot:

http://ds4windows.com/images/p1.png

However, with DS4Windows, this option currently only works for wired USB and not for wireless Bluetooth unfortunately, see:

https://github.com/Jays2Kings/DS4Windows/issues/48

My1, 2016-01-09:
the point is that you could for example change the lightbar color where I still have found no software that does that without interfering...

kq01526, 2016-01-09:
@nefarius :

What @My1 is saying is another good point. ScpToolkit has the ability to configure the DS4 lightbar and buttons and so on. If you have to uninstall ScpToolkit to switch to the DS4's native DirectInput mode, you lose all this configurability.

Adding an option to ScpToolkit to switch from the emulated Xbox controller XInput mode to the native  DS4 Direct Input mode would solve all of this, it would give you the configurability and at the same time would allow to play games which support it to run with the native DS4 Direct Input mode and to see PlayStation button prompts instead of Xbox button prompts (which is a huge plus).

nefarius, 2016-01-09:
Ooooohh I understand now. Hm, well that doesn't sound too bad, any plans from DS4Windows in that direction?

kq01526, 2016-01-09:
> @nefarius wrote:
> 
> any plans from DS4Windows in that direction?

I've already answered that question half an hour ago, see:

> @kq01526 wrote in https://github.com/nefarius/ScpToolkit/issues/89#issuecomment-170268284 :
> 
> [DS4Windows](http://ds4windows.com/) also has integrated an option to switch to DirectInput, see the "Use DInput only" checkbox in the lower right corner of the following screenshot:
> 
> http://ds4windows.com/images/p1.png
> 
> However, with DS4Windows, this option currently only works for wired USB and not for wireless Bluetooth unfortunately, see:
> 
> https://github.com/Jays2Kings/DS4Windows/issues/48

:stuck_out_tongue_closed_eyes:

My1, 2016-01-09:
well for BT there is yet nothing because the Dinput mode doesnt work with BT yet, but it does over USB (for some or another reason) but personally I'd rather use DS4 windows because it also serves as a nice replacement for JoyToKey

nefarius, 2016-01-09:
I have/had no plans to somehow compete with DS4Windows so I don't know what to say honestly.

My1, 2016-01-09:
yeah of course, that was just my opinon. but if there is a software that does those "extra features" of e.g. changing the lightbar and similar stuff without meddling otherwise I'll still gladly use it.

kq01526, 2016-01-09:
> @My1 wrote:
> 
> [...] without meddling otherwise [...]

Just to clarify: When he says "without meddling otherwise" he/she means "without emulating a Xbox XInput controller".

nefarius, 2016-01-09:
@My1 from my understanding of the DS4 this will not be possible with DS4Windows without some rather nasty hacks. DS4Windows communicates with the DS4 in its "PC mode" where it simply presents itself to the PC as a HID device. Manipulating extra stuff like the Lightbar can only be done when the DS4 is in "PS4 mode", which is what Scp uses. I don't think these modes are "live switchable" and compatible with the common Bluetooth stacks and Windows HID drivers.

My1, 2016-01-09:
but when this thing is in PS4 mode why do we still see a "frozen" HID DS4 in devices and printers

(frozen=buttons and stuff do nothing)

nefarius, 2016-01-09:
@My1 hm? Under which circumstances? Btw. [should we switch over to the chat](https://gitter.im/nefarius/ScpToolkit) since we're hijacking this issue thread quite a bit? ;)