nefarius/ScpToolkit #90: "Fake" Controller works with cable, shows as connected via BT, yet it's not

starg09, 2015-12-15:
All right, so apparently I just purchased a fake DS3. I'm totally fine with it, as it works on the PC as I intended. Well, at least on cable.

I get a "Fake controller" warning on connection, but it works perfectly. When I try to connect it through Bluetooth, though, I get a "Fake controller, trying workaround 1" warning, a "Connected on virtual port 1" message, yet it doesn't seem to work (neither does the controller mark it being connected as joystick 1.

Not sure what specific fake model it is, all I know is that it originally said "PS(R) Controller" as the device, and didn't work at all before using SCP. I have Windows 10 Pro T2, 64-bit.

I attach the debug info, and the notifications I get when connecting via bluetooth (that is, pressing the PS button with the dongle connected and configured)

[image]

[The debug info](https://mega.nz/#!M5EHUYIT!ZbhkEkTZXpSsV0qvYeY9YAjA0658xweFxrSVgVYcd6U) (GitHub won't let me upload the zip for some reason)

nefarius, 2015-12-15:
Your service log is damaged and can't be read. Please repeat the connection procedure and shut down the service before you run the debug collector!

starg09, 2015-12-15:
Ok, one second.

starg09, 2015-12-15:
https://mega.nz/#!UlsxwBYT!_9rFgCwTgtZFOquRduxoj7QLAg2_JjI2m9L_xHa4kJM

nefarius, 2015-12-15:
The error is still there, please stop the service, go to your installation directory, delete the file `Logs\ScpDsxService_GUILLE7$.log.xml` and repeat the steps above.

starg09, 2015-12-15:
All right. Maybe it's because I'm on a limited account (with access to admin, but still)

starg09, 2015-12-15:
https://mega.nz/#!E1lzxSLI!7s-4Xnvdd0sb7zdq8R0WEIe6rv3aKkvtuO5fcUJIuUk

nefarius, 2015-12-15:
It is! But could you also make a screenshot of ScpToolkit Monitor while it's supposedly connected via Bluetooth?

starg09, 2015-12-15:
Connected via cable: http://i.imgur.com/L9WmxRu.png

Once disconnected: http://i.imgur.com/fGC0Bvu.png

Once "Connected": http://i.imgur.com/6HKPs11.png

starg09, 2015-12-15:
On a side note: I forgot to check that before, but I had read something about that "0000 None" error. Although it might not be even important, I tried stopping the service, opening it through the standalone version, and once in that state checked if the "Vibration" buttons worked. They don't.

nefarius, 2015-12-15:
Mystery solved; these zeros are the packet counter; if it stays at zero that means the data the controller is sending can't be interpreted. For the moment you're simply out of luck, sorry.

starg09, 2015-12-15:
Oh, okay. Thanks for checking anyway...

At least it works with cable, it's something :P

nefarius, 2015-12-15:
I will come back to the counterfeit ones but just no time at all right now =/

starg09, 2015-12-16:
Sounds great to me. No hurries, thanks for working on this! :D

angelflames1337, 2015-12-17:
I am having the exact same issues. Not to push you or anything, but will the fake support come within these 2 months? Coz if it is, I am preordering a game tomorrow to use this :p

Koeleskab, 2015-12-18:
I have the exact same issue on USB with what turns out to be a counterfeit DS4 controller. Should i send along the logs, in case the time ever comes for the lesser issues?

xorincret, 2015-12-23:
@nefarius 

Same issue here, fake DS3.

Pairing should be OK since the PC and controller can communicate enough to determine it's a 3rd party controller. I would assume that the controller's interpretation of the protocol is also fairly compliant given that it's marketed for use with the original PS3.

Any logs I can provide that would give you a head start on fake controllers, or this one specifically?

I had a look at the logs in the generated zip but they didn't have much useful information IMHO. Perhaps I don't know what to look for. Is there a feature in ScpServer that can be activated to save BT packets to disk? This should be useful for debugging, to have users with fake controllers send their packet dump.

I don't mind putting in a few hours of coding if it would help the project advance.

nikhosagala, 2015-12-25:
I've some issue here.

nefarius, 2015-12-28:
@xorincret: I got some different "breeds" of those PANHAIs donated a few month ago and had them disassembled. I managed to tap into the communication between the Bluetooth controller chip and the micro controller (they communicate via UART thankfully) and actually could sump the streams they exchange when connecting to a genuine Sony PlayStation 3. 

Since my knowledge of the whole Bluetooth specification (if I recall correctly, v4 has nearly a 1000 pages...) is very limited I wasn't yet able to decipher those dumps. I don't have much time either since I'd like to focus on development rather than reverse engineering. 

The Logs the toolkit produces won't help much since the inner workings of the protocols involved aren't logged since it would add too much of an overhead.

What I've discovered so far is, that the **actual** pairing (sadly people misuse this term all the time; the flashing LEDs do not necessarily indicate a pairing problem) works fine, it's the L2CAP connections which fail after responding to a connection and configuration request. The PANHAIs just don't send the 3rd connection request for the HID Service channel which the originals do just fine. That's where the software is currently stuck.

If you want to know more about the inner workings of the Scp Bluetooth stack don't hesitate to ask.

luzair, 2015-12-30:
@nefarius I have the same issue "Fake controller detected". Should I ask for a refund from the supplier? The supplier told me its a genuine sony controller. I was wondering why old version of motioninjoy and scpserver works with bluetooth, but ur latest version doesnt work with bluetooth on these fake controllers?

When using old scpserver, I can get my controllers to vibrate on usb mode, but not on bluetooth mode. PS. I can connect via bluetooth using old scp server method to this fake controllers.
Please advise.

starg09, 2015-12-30:
@luzair not nefarius but you should. I did, same situation (sold as
original), mentioned that seemed to not be the case and I was almost
instantly offered "a replacement or a refund". Who knows, maybe those fancy
camouflage ones they sell are actually original, or they offer a dualshock
4 (I can dream hehe :P ) (seriously, I'll most likely get refunded)

So, long story short: you should ask for a refund xD

luzair, 2015-12-30:
@starg09 Thanks for the advise, I will contact my supplier asap. What about rumble support for old scp version. using SCP-DS-Driver-Package-1.2.2.175. I can get it to work via bluetooth but with no rumble support.

aking4125, 2015-12-30:
I can confirm the fake controllers working with "SCP-DS-Driver-Package-1.2.2.175" via bluetooth flawlessly. I don't understand they don't work in this new toolkit thing. Both works with USB though.

luzair, 2015-12-30:
@aking4125 When u use Fake DS3 and SCP-DS-Driver-Package-1.2.2.175 via bluetooth, do u have rumble or vibration while playing games?

aking4125, 2015-12-30:
@luzair Nope. But I don't really care. So long as it works. Though now out of nowhere, 1 of the controllers is not being detected and it refuses to install the controller drivers. Then I BSOD'd. This is getting crazy.

luzair, 2015-12-30:
@aking4125 I think ur controller's battery is burnt. 
I remember having rumble via bluetooth when i used motioninjoy. but im not sure if I should use MIJ again. Any suggestions?

aking4125, 2015-12-30:
The vibration works perfectly fine otherwise. My controllers is fine as they work via USB. 

What I'm trying to say, The SCP-DS-Driver-Package-1.2.2.175 was the only way to get them to work via Bluetooth. This SCP Toolkit, doesn't work with bluetooth. Both works with USBs. So since the other one worked. I used that and everything was fine until I restarted my PC. It didn't restart properly, it BSOD'd with the SCPVBUS error or something. Restart. Now the old software refuses to install the controller drivers. I cannot gain access to my system32 folder to manually copy them anymore. 

Trying the toolkit again and it looks like I have to reinstall Windows.

luzair, 2015-12-30:
Yes, I really don't wana use MotioninJoy, I will still with SCP-DS-Driver-Package-1.2.2.175 then.

@aking4125 You can try uninstallling relavant softwares one 1 by then restart. Then try clean registry with CCleaner. I know formatting windows is a pain.

I hope somebody can tell me, how to make fake DS3 have rumble via bluetooth.

aking4125, 2015-12-30:
None of that worked. It can't seem to gain access to the system32 folder. I found a way for me to do it. Now the SCP-DS-Driver-Package-1.2.2.175 doesn't work with USB. BUT! Bluetooth works flawlessly again! Thank god. That's all I wanted was the controllers to function. Vibration doesn't work like this but it's not the enough of the world for me. 

Now, SCP Toolkit, won't work with Bluetooth. Period. I don't understand why though.

luzair, 2015-12-30:
@aking4125 I don't know what are you doing to ur poor pc haha

The only person who knows why SCP Toolkit won't work with bluetooth is @nefarius.

luzair, 2015-12-30:
Do you think we should like Better DS3?

aking4125, 2015-12-30:
I know what I'm doing as I manged to get it to work. Now my family can play Viscera Clean-Up Detail wirelessly with our PS3 controllers

Better DS3 requires the normal DS3 tool drivers, which means you need MotioninJoy. Don't use it. For now, if users want bluetooth on some crap 3rd party PS3 controllers. If SCP ToolKit doesn't work, use the old SCP-DS-Driver-Package-1.2.2.175 for that. Otherwise just use SCP Toolkit. 

I think it'd may be best to just get Xbox 360 controllers or Playstation 4 controllers. Even though they are much more expensive than the cheap PS3 controllers (I got the four I'm using for $9 each). Because with those, especially the PS4 controller. They just work without having a 3rd party Bluetooth driver. And DS4 is easier to use. Too bad that doesn't work with PS3 controllers.

I bid @nefarius luck with whatever the issue is. Cause in SCP-DS-Driver-Package-1.2.2.175 Bluetooth works, just no vibration which I don't mind at all.

angelflames1337, 2015-12-30:
Aking, unfortunately SCP-DS-Driver-Package-1.2.2.175 doesnt work for me. I click the scpdriver.exe and nothing happened. Check the task manager and it seems the exe appear for a few seconds then it gone. Tried to unblock, zip/unzip and few other thing, nothing works. I just want a bluetooth function on my ps3 controller T_T

aking4125, 2015-12-30:
@angelflames1337 You installed it incorrectly. You have to load the 1.2.0 files first. then the 1.2.2 files and replace everything. That's because the 1.2.2 is just an update. The core files is within' the 1.2.0 archive.

This doesn't explain however why the Toolkit doesn't work in Bluetooth mode. But the old 1.2.2 one does.

angelflames1337, 2015-12-30:
Ah I see, I'll try that and get back here :)

aking4125, 2015-12-30:
This isn't the thread or place to be talking about this. Perhaps another time via a PM or something.

nefarius, 2015-12-30:
@aking4125 the reason for my fork behaving differently than the original Scp drivers is simple: I **hate** hacks where even if you kind of understand what the code does you still have no idea why it works. I ripped out a good bunch of those mysterious sections (beware that the original code base was poorly documented too) so I could **continue** working in refining and enhancing the software without pissing genuine owners off because the "fake workarounds" can (and did) break connectivity with Sony controllers. So I'm simply playing safe until I was able to fully understand and reverse engineer the counterfeit ones to code a *real* solution that's reliable and stable.

I hope this answer clears things up for you.

luzair, 2015-12-31:
@nefarius I would like to make a personal request to you. Hope to get a favorable reply. 

Currently I'm using fake controllers with SCP-DS-Driver-Package-1.2.2.175. When using USB, vibrations work, but when using bluetooth vibrations don't work. Please help to get vibrations to work via bluetooth. It will benefit  many, me included, who got deceived by sellers who sold these fake controllers as the authentic ones. I know you can do it and hope you will help us.

aking4125, 2015-12-31:
@nefarius Yes that makes sense. Well, until you do. Users such as myself cannot use SCP Toolkit for Bluetooth with some PS3 controllers. I tried I genuine Sony PS3 controllers from a friends PS3. That still didn't work via Bluetooth with the Toolkit. The 1.2.2 older drivers does for some reason.

@luzair The old SCP 1.2.2 drivers is not under nefarlus at all. That one is no longer supported. So you are stuck without Vibration via Bluetooth. I honestly don't even think that's an issue. So long as the controllers work then just leave it be. But until nefarlus can figure out the issue with the Bluetooth in the current SCP Toolkit. There is nothing can really be done. As seeing how our issue seems to be very specific. It may never happen. 

I'm just going to save up what little money I have and get 3 more legit PS4 controllers and use DS4. I do want to use my Bluetooth adapter for other stuff and with the PS3 controllers I can't really do that which for now it isn't an issue to me. Or I should just buy another Bluetooth adapter for everything else.

luzair, 2015-12-31:
@aking4125 The most cost effective option for you is to buy another bluetooth dongle. After going so much hassle with ps3 controllers, next time im buying an xbox360 controller. hope they are legit as they claim to be. sometimes i think those third party gamepads that can emulate x360 profile out of the box are much more hassle free. Happy New Year every1.

aking4125, 2015-12-31:
Yeah. But I do play PS2 and PS1 games and playing them with an Xbox 360 controller is annoying you know? 

Besides, the Xbox controllers usually require batteries and a rechargeable pack is just extra money. I'll eventually get them all. And I will get another Bluetooth dongle. I'm aware. I'm still gonna get 3 more PS4 controllers though cause this was more hassle than it needed be xD

aking4125, 2015-12-31:
@angelflames1337 If even the old SCP 1.2.2 doesn't work for you in Bluetooth. Either make sure the Bluetooth Drivers and DS3 drivers are properly installed or you're Bluetooth Dongle is not supported.

If the installation reads it cannot write the files to the system32 folder. You need extra permissions to get that working. You need to select the .exe. Make sure to run it as an Admin. And then, go to the special Permissions and allow your current Admin account Effective access. That's what I had to do. After that, the SCP Service will take over and that worked for me.

Just wished it worked with the toolkit from the get go but oh well.

luzair, 2015-12-31:
Whats the difference between the toolkit version and the old SCP 122 version, apart from supporting fake controllers.

luzair, 2015-12-31:
@aking4125 You should try third party controllers, they can interchange between ps3 controller profile and x360 controller profile with just a press of a button. BUT they only last for a week. haha. I dun mind using them, if they were longer lasting and had rechargeable batteries.

aking4125, 2015-12-31:
@luzair If you read my first post (Or any of them). I am using 3rd party Controllers. The "Fake" controllers. The Difference with the Toolkit and 1.2.2 is that the the Toolkit doesn't work for me via Bluetooth and the 1.2.2 one does as stated multiple times above. 

If you mean, real difference. Toolkit is easier to install, works with PS4 controllers, has profile settings on it's own. More options and is more sleeker.

luzair, 2015-12-31:
@aking4125 Yes, I know, we're using fake controllers, me included, (i got deceived by seller that these controllers are legit, until I used toolkit). I know fake controllers work with bluetooth using 1.2.2 but not with toolkit. I mean other differences. In the toolkit, I noticed that we can assign battery status in the 4 leds instead of showing player 1. When I mean 3rd party controllers, i mean those that don't even look like ps3 controllers. Those that have turbo and / or auto fire functions sometimes.

angelflames1337, 2016-01-01:
@aking4125 Holy shit my controller finally works via bluetooth. After installing, apparently I just need to press the back reset button using a needle, and it just seem to pair afterwards. Thanks mate!

xendeus, 2016-01-30:
I hope PANHAI gamepad will work now. im still waiting for an update

luzair, 2016-02-12:
Anybody knows how to have rumble support on fake controllers using SCP-DS-Driver-Package-1.2.2.175? Any help will be appreciated.

aries-soft, 2016-02-23:
Download the moded ScpControl.dll from here, the dll was built for such devices.
Extract ScpControl.dll from the downloaded archive
Replace the ScpControl.dll in your bin folder with this.

http://forums.pcsx2.net/Thread-XInput-Wrapper-for-DS3-and-Play-com-USB-Dual-DS2-Controller?pid=444231#pid444231

luzair, 2016-02-23:
@aries-soft Can you confirm if the controller would work with vibration over bluetooth?

aries-soft, 2016-02-23:
Works, but sometimes the vibration disappears

luzair, 2016-02-24:
yup, tried it last nite, still no vibration. is there any additional drivers for the vibration?

aries-soft, 2016-02-24:
Vibration via bluetooth only works with the new adapters (eg from 4.0 works there with the old 2.0)

luzair, 2016-02-25:
ok. that gives me more variables for me to experiment on. will let you guys know, if i ever try it.

luzair, 2016-02-25:
is there any app that tests the vibration on the controllers? it doesn't have the test vibration option in the 'game controllers' by windows.

charlesmilette, 2016-02-25:
Any game that has a vibration feature may do the trick :)

[This link](http://www.getdare.com/bbs/showthread.php?t=146897) claims to say that you can test it with x360ce, but I haven't tried it.

VarunBurde, 2016-03-06:
Hello,
@luzair @starg09 @aking4125 
Guys have you tried ScpServer_v1.5.5749.18036 .
Its working for me along with rumble.
https://github.com/nefarius/ScpToolkit/releases/tag/v1.5.5749.18036
try this link 
and thanks @nefarius for your great work.

Fry1985, 2016-03-10:
@nefarius
The last release is from 10 Jan...
are u still working on the Problems?
When will fake Controllers probably be supported?
Im looking forward to that.
Great Work till now with this tool.
Keep it on. ;-)

@aries-soft
will the modded scpcontrol.dll work with SCP Toolkit or only with SCP DS3 Driver?
Does this modded dll make Fake controllers work? U said something about Rumble?
Thats not what this Thread is for.
http://forums.pcsx2.net/Thread-XInput-Wrapper-for-DS3-and-Play-com-USB-Dual-DS2-Controller?pid=444231#pid444231

aking4125, 2016-03-10:
@VarunBurde I have not. Will give it a shot when I'm free from IRL stuff. Will report back within' 24Hr

nefarius, 2016-04-06:
@Fry1985 
I know.
Yes.
Don't know.
Great.
Thanks.
I try =]

[17 more replies]

closed 2018-01-26 by nefarius